turtlebot/turtlebot4 #99: TurtleBot 4 Not Showing Up in Rviz/Map Not Updating During SLAM

BashByr, 2023-01-27:
I recognize this is a similar to Issue #44, however I believe this is a different case since the Create3 appears to be connected to the same network as the RPI and my computer (ros2 topics shown below). I will attempt to detail this as much as I can, from startup to the source of error.

First, to check connectivity to the robot, I run `ros2 topic list`, and the following is the output:

```
bash@ubuntu:~$ ros2 topic list
/battery_state
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/imu
/interface_buttons
/ip
/joint_states
/joy
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/camera_info
/stereo/depth
/tf
/tf_static
/wheel_status
```
 
Running `ros2 node list` returns the following:
```
bash@ubuntu:~$ ros2 node list
/analyzers
/rgb_stereo_node
/robot_state_publisher
/rplidar_composition
/turtlebot4_diagnostics
/turtlebot4_node
```

Please also note that I can ping my RPI from my computer as well as SSH into it successfully.

Now, running `ros2 launch turtlebot4_navigation slam_sync.launch.py`, I almost instantly get the following messages:

[image]

Then, when I run `ros2 launch turtlebot4_viz view_robot.launch.py` in a separate terminal, I get the following output in the terminal:

[image]

And the Rviz window looks like this:

[image]

It gets a correct initial lidar scan  but fails to keep updating it as the robot moves around. Further, the robot is nowhere to be seen on the map. Running `ros2 topic list` as SLAM is running generates the following output:

```
bash@ubuntu:~$ ros2 topic list
/battery_state
/clicked_point
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/hazard_detection
/imu
/initialpose
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/kidnap_status
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_metadata
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/particle_cloud
/plan
/robot_description
/robot_state/transition_event
/rosout
/scan
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/waypoints
/wheel_status
/wheel_ticks
/wheel_vels
```

Could I please get some pointers as to what may be wrong? Any help is appreciated, thanks!

roni-kreinin, 2023-01-27:
@BashByr Can you try running asynchronous slam on the RPi?

 `ros2 launch turtlebot4_navigation slam_async.launch.py`

BashByr, 2023-01-28:
That worked, thank you! I was able to build a map and use Nav2 Goal in Rviz successfully. Any idea why synchronous slam doesn’t work?

roni-kreinin, 2023-01-30:
Closing as issue is resolved.

closed 2023-01-30 by roni-kreinin

Basavaraj-PN, 2023-02-27:
I'm getting the same issue, not able to run anything, tried with asyn launch file.