turtlebot/turtlebot4 #101: RViz shows nothing

hamidosooli, 2023-01-30:
**Please provide the following information:**
 - OS: Ubuntu 20.04.5-desktop-amd64
 - ROS Distro: Galactic
 - Built from source or installed: installed according to the official [documentation](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html)
 - Package version: 
 sudo dpkg -s ros-galactic-turtlebot4-navigation
Package: ros-galactic-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 388
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 0.1.2-1focal.20221208.082815
Depends: ros-galactic-nav2-bringup, ros-galactic-nav2-simple-commander, ros-galactic-slam-toolbox, ros-galactic-ros-workspace
Description: Turtlebot4 Navigation

 **Expected behaviour**
 A view of the generated map and the robot should be available in the RViz.

 **Actual behaviour**
[image]
As it is shown in the figure, RViz shows nothing!

**To Reproduce**
Provide the steps to reproduce:
I'm trying to generate a map of the environment using the steps mentioned in the [manual](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html). 
 
_To make sure that the RPLIDAR and description nodes are running on the TurtleBot 4 I ssh into the robot and run the following commands:
ros2 launch rplidar_ros rplidar.launch.py
ros2 launch turtlebot4_description robot_description.launch.py

_Then I open a new CLI, and source my environment (source /opt/ros/galactic/setup.bash), and export the relevant ROS_DOMAIN_ID, ( export ROS_DOMAIN_ID=1, he ros domain ID of the robot is similar. In the ssh'd cli, echo $ROS_DOMAIN_ID shows 1 as well). Following the manual, I run the sync slam by the following command:
ros2 launch turtlebot4_navigation slam_sync.launch.py
In another cli after running similar commands I use the following command to run RViz (it runs to the mentioned messages)
ros2 launch turtlebot4_viz view_robot.launch.py
[INFO] [launch]: All log files can be found below /home/hamid/.ros/log/2023-01-30-16-54-54-367145-hamid-Alienware-Aurora-R12-23625
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [23627]
[rviz2-1] [INFO] [1675115694.729207981] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1675115694.729304192] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-1] [INFO] [1675115694.759640102] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [WARN] [1675115694.831144657] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1675115694.832325717] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1675115694.833558178] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.

**Other notes**
_I have two wifi networks (2.4 GHz and 5GHz) provided by the same router and bridged to each other. Raspberry Pi is connected to my 5GHz network and Create3 is connected to my 2.4 GHz network. The LCD and all of the green leds on the robot are on and I can ssh into the robot.
_Turning the ubuntu firewall off, does not affect this issue
_Information about nodes and topics on the robot is provided below

ubuntu@ubuntu:~$ ros2 node list
WARNING: Be aware that are nodes in the graph that share an exact name, this can have unintended side effects.
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/analyzers
/joy_linux_node
/motion_control
/rgb_stereo_node
/robot_state
/robot_state_publisher
/robot_state_publisher
/rplidar_composition
/rplidar_composition
/static_transform
/system_monitor
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
/ui_mgr

ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/system_monitor/transition_event
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

roni-kreinin, 2023-01-31:
You don't need to launch the rplidar and robot description because they already get launched by the turtlebot4 bringup launch file. 

Can you paste the output from the SLAM launch?

Also, can you try switching the Fixed Frame under Global Options to 'odom' in Rviz? You should be able to see the robot model.

hamidosooli, 2023-01-31:
Here is the output from the SLAM launch
hamid@hamid-Alienware-Aurora-R12:~$ ros2 launch turtlebot4_navigation slam_sync.launch.py
[INFO] [launch]: All log files can be found below /home/hamid/.ros/log/2023-01-31-13-35-19-803704-hamid-Alienware-Aurora-R12-7121
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [7123]
[sync_slam_toolbox_node-1] [INFO] [1675190119.889286128] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1675190119.898008405] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1675190119.898083966] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.

and here is the Rviz after changing the fixed frame to odom
[image]

roni-kreinin, 2023-01-31:
Do you see the same topics calling `ros2 topic list` from your PC as you saw on the RPi?

hamidosooli, 2023-01-31:
I just see the following:

hamid@hamid-Alienware-Aurora-R12:~$ ros2 topic list
/parameter_events
/rosout

hamidosooli, 2023-01-31:
No I just see the following:

hamid@hamid-Alienware-Aurora-R12:~$ ros2 topic list
/clicked_point
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/initialpose
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_metadata
/map_updates
/mobile_base/sensors/bumper_pointcloud
/parameter_events
/particle_cloud
/plan
/robot_description
/rosout
/scan
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/tf
/tf_static
/waypoints

roni-kreinin, 2023-01-31:
Have you called `export ROS_DOMAIN_ID=1` on your PC? And are you on the same Wi-Fi network as the turtlebot?

hamidosooli, 2023-01-31:
yes

roni-kreinin, 2023-01-31:
On your PC can you call `ros2 multicast receive`, then in a separate terminal SSH into the RPi and call `ros2 multicast send`? Your PC should receive a packet from the RPi.

hamidosooli, 2023-01-31:
I did this on the robot

ubuntu@ubuntu:~$ ros2 multicast send
Sending one UDP multicast datagram...

However I think the PC has an issue communicating to the robot. It just gets stuck on the following:

hamid@hamid-Alienware-Aurora-R12:~$ ros2 multicast receive
Waiting for UDP multicast datagram...

roni-kreinin, 2023-01-31:
You may have a firewall or some other router setting that is blocking multicast packets. Try looking through your router settings.

iRobot just released the new G.4.1 firmware for the Create 3 which adds support for the [discovery server](https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html). I am working on instructions for setting that up, as well as a new galactic image with this feature. This networking method lets us disconnect the Create 3 from Wi-Fi, and use only the 5GHz WiFi of the RPi, and may also help fix multicast communication issues like yours.

hamidosooli, 2023-01-31:
see the following

^Chamid@hamid-Alienware-Aurora-R12:~$ sudo ufw disable
[sudo] password for hamid: 
Firewall stopped and disabled on system startup
hamid@hamid-Alienware-Aurora-R12:~$ ifconfig
enp4s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.16.33.129  netmask 255.255.254.0  broadcast 172.16.33.255
        inet6 fe80::dd50:35d2:86f5:f4a4  prefixlen 64  scopeid 0x20<link>
        ether 30:d0:42:fe:78:85  txqueuelen 1000  (Ethernet)
        RX packets 9116  bytes 2968995 (2.9 MB)
        RX errors 0  dropped 3745  overruns 0  frame 0
        TX packets 4764  bytes 1133762 (1.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 59652  bytes 12189389 (12.1 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 59652  bytes 12189389 (12.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp5s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.50.54  netmask 255.255.255.0  broadcast 192.168.50.255
        inet6 fe80::cf5d:6c82:fb54:683b  prefixlen 64  scopeid 0x20<link>
        ether 10:6f:d9:a8:60:eb  txqueuelen 1000  (Ethernet)
        RX packets 10527  bytes 3687517 (3.6 MB)
        RX errors 0  dropped 1070  overruns 0  frame 0
        TX packets 5427  bytes 1028333 (1.0 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
Firewall is disabled and the MULTICAST is active.

roni-kreinin, 2023-01-31:
This is on your PC, but your Wi-Fi router may have its own firewall and multicast settings.

hamidosooli, 2023-01-31:
Thanks I tried with a new router and network and everything worked well

closed 2023-01-31 by hamidosooli