turtlebot/turtlebot4 #103: Invalid frame ID "wheel_drop_right" passed to canTransform argument source_frame - frame does not exist

rgariepy, 2023-02-05:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed
 - Package version: Unsure where the offending package is

 **Expected behaviour**
Frames wheel_drop_right and wheel_drop_left exist

 **Actual behaviour**
Frames do not exist

**To Reproduce**
1. Turn on TBT4 Standard, wait for bringup
2. Run ros2 launch turtlebot4_viz view_model.launch.py

Or, run view_frames and note lack of frames

**Other notes**
iRobot firmware version is G.4.1
Can view other Create3 topics (ex. /hazard_detection) and RasPi topics (ex. /scan) from the computer in question
On home wifi with no fancy filtering going on

Full topic list:
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

rgariepy, 2023-02-05:
[frames.pdf](https://github.com/turtlebot/turtlebot4/files/10612001/frames.pdf)
TF tree

roni-kreinin, 2023-02-06:
This looks like the same issue as https://github.com/iRobotEducation/create3_sim/issues/125. It's been stale for a while so I will ask iRobot to look into it again.

roni-kreinin, 2023-02-16:
The issue has been fixed in `humble`, likely won't backport to `galactic` as it is EOL.

The issue was basically that those links are connected to prismatic joints in the URDF and the `robot_state_publisher` does not publish static tf's for those joints. The fix is just adding a static transform publisher.

closed 2023-02-16 by roni-kreinin

rgariepy, 2023-02-16:
Can that be added to the tutorials?

roni-kreinin, 2023-02-21:
The static transform publishers were added to the [description launch](https://github.com/turtlebot/turtlebot4/blob/humble/turtlebot4_description/launch/robot_description.launch.py#:~:text=left_wheel_drop_stf%20%3D%20Node), so there won't be anything for the user to do in Humble. I can add it to the FAQ for galactic though.

redchillipadi, 2023-05-18:
I am also tracking this down, I think the relevant commit is 7f6d8f0