turtlebot/turtlebot4 #107: Not getting odom imu topic

Basavaraj-PN, 2023-02-14:
After exporting ROS_DOMAIN_ID=0, I'm not able to see few topics in my laptop except below listed.

```
➜  ~ ros2 topic list 
/battery_state
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/camera_info
/stereo/depth
/tf
/tf_static
/wheel_status
```

Basavaraj-PN, 2023-02-14:
Getthing these inside RPI.

```
ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels
```

roni-kreinin, 2023-02-14:
Have you connected the Create 3 to a bridged 2.4GHz network? Also have you applied the cycloneDDS profile as mentioned [here](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#network-configuration)?

Basavaraj-PN, 2023-02-14:
Yes, I'm on 2.4Ghz band and also DDS profile is set as per the document.

Basavaraj-PN, 2023-02-17:
tried still no luck.
How to get back AP mode again? what is the button combination?

closed 2023-02-23 by Basavaraj-PN

NikoShawn, 2025-08-25:
refer this:https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html