turtlebot/turtlebot4 #108: Raspberry Pi's WiFi configuration does not finish

languyen14507, 2023-02-14:
The command
sudo wifi.sh -s '<WIFI_NAME>' -p '<PASSWORD>' -r 'USA'
caused the Turtlebot4 terminal to hang up.

roni-kreinin, 2023-02-14:
@languyen14507 The terminal will hang because the RPi has switched to a different network, so your SSH session will no longer work. Switch your PC to the network that you connected the RPi to and try to echo the `/ip` topic.

Also, if you did not add call  `sudo reboot`, you may need to power cycle the RPi for the ROS2 node to start up properly.

languyen14507, 2023-02-15:
Since the SSH terminal hang, I am not sure if the Turtlebot 4 WiFi configuration was done properly. I tried to power cycle the RPi (power off the robot then power it back on) then run the command
ros2 topic echo /ip 
and got the following

1676478586.782825 [0]       ros2: wl01: does not match an available interface supporting udp
[ERROR] [1676478586.782888566] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:261'

with this new error message:

  'rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:413'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1676478586.782946019] [rcl]: Failed to fini publisher for node: 1
error creating node: rcl node's rmw handle is invalid, at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/node.c:413

Is there a way to reconfigure the RPi's WiFi network?

roni-kreinin, 2023-02-15:
That looks like an issue with your cycloneDDS configuration on your PC. `wl01` doesn't look like a standard WiFi interface name. Can you check your interfaces by calling `ip a`?

languyen14507, 2023-02-15:
languyen:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eno2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 04:d4:c4:e6:47:09 brd ff:ff:ff:ff:ff:ff
    altname enp3s0
3: wlo1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 08:71:90:66:fc:bf brd ff:ff:ff:ff:ff:ff
    altname wlp0s20f3
    inet 192.168.11.142/24 brd 192.168.11.255 scope global dynamic noprefixroute wlo1
       valid_lft 84159sec preferred_lft 84159sec
    inet6 fe80::bfd4:3c87:beb8:7f99/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

languyen14507, 2023-02-15:
Sorry, I mistyped. It was supposed to be wlo1 and not wl01. The "ros2 topic echo /ip" command does work and print the robot IP address on the terminal. Thank you.

roni-kreinin, 2023-02-15:
Good to hear that its working. I will close the ticket now, feel free to re-open if you run into more issues, or make a new ticket.

languyen14507, 2023-02-25:
Hello,
I am having another problem with my Turtlebot 4. For some reason, the onboard display where I used to see the bot's IP address and undock, dock,.. selections is completely off. I can connect to the Create@3 via the Create-xxxx net work and verify that the Create@3 is publishing its topics
...
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

but I can't drive the robot because I can't undock it. I tried to also use the teleop_twist_keyboard node but it doesn't work either. Please advise. Thank you.