turtlebot/turtlebot4 #109: Using Turtlebot4 Software for Custom Hardware

4DJR, 2023-02-14:
I currently have a custom input that is publishing values to the /cmd_vel topic that is able to move the Turtlebot around. I would like to now make it so that I can also control another set of motor drivers that are unrelated to the Turtlebot but use the same /cmd_vel topic so that both the other set of motors and the Turtlebot are moving at the same time using the same input. Is there a file that I can copy and modify for the new set of motors so that I can assign the appropriate pin numbers and send the correct PWM to my other motor drivers? How much of the file will I need to customize? Do I need to change wheel size, track width, etc. somewhere for it to work? Where does the Turtlebot convert the /cmd_vel topic into a PWM signal to move its own motors? Thanks.

tonybaltovski, 2023-02-14:
Hello!

Great questions! Unfortunately this isn't quite the correct place for it. The iRobot Create3 does the `cmd_vel` to PWM conversion internally and the source code is not available. [Here](https://www.youtube.com/watch?v=-PCuDnpgiew&ab_channel=ArticulatedRobotics) is video that should help. Your best bet would be to ask the ROS answers community, (https://answers.ros.org/questions/) there's usually plenty of ROS experts there that will be willing to help.

Best of luck!

closed 2023-02-14 by tonybaltovski

4DJR, 2023-02-14:
Thanks for the links, I'll check those out. I was just wondering, is it possible to change the parameters of the Turtlebot to use my motors instead? Is there a config file that I can just change the PWM pins somewhere or something like that?

tonybaltovski, 2023-02-15:
No, it is not.  The iRobot Create3 is an integrated unit that has specific motors that interface with the embedded controller.