turtlebot/turtlebot4 #110: OAK-D Camera Works Intermittently

4DJR, 2023-02-14:
**Please provide the following information:**
OS Image: Using Ubuntu 20.04 ROS2 Galactic Turtlebot4 Lite image (0.1.3) provided from 
http://download.ros.org/downloads/turtlebot4/

 **Expected behaviour**
I am expecting the OAK-D camera to display the image when runnning Rviz2

 **Actual behaviour**
Sometimes the camera will work normally when the Turtlebot is initially turned on and I can see the image without issue. However, eventually the camera will stop displaying the image and even after unplugging and plugging it back in it does not return. Sometimes it also does not show up when the Turtlebot is initially turned on. It just seems like it works unreliably. I have tried using the provided camera cable and also several others with no improvement.

Any idea why the camera won't stay on as expected? How can this be fixed?

Thanks.

roni-kreinin, 2023-02-14:
Can you call `sudo journalctl -u turtlebot4 -r` and see if the oakd node is showing any errors?

4DJR, 2023-02-15:
So I rebooted the turtlebot this morning and everything was mostly working fine. I could see the image using the camera display in rviz, however, I could not see the image when using the image display. Now, later in the day, the camera is not working at all. Using that command you suggested I can see something regarding the rgb_stereo node but it looks like it's just defaulting to a different value and doesn't seem like it's crashing. Here's the output of "sudo journalctl -u turtlebot4 -r" since the last reboot when the camera display stopped showing anything:

Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [rgb_stereo_node-5] [1844301031E2E80F00] [1.1.2] [40.865] [MonoCamera(0)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [rgb_stereo_node-5] [1844301031E2E80F00] [1.1.2] [40.864] [MonoCamera(1)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.384952407] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.384933472] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Mouse'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.384859178] [GenericAnalyzer]: Analyzer 'Mouse' matches 'turtlebot4_diagnostics: /mouse topic status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.383183705] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.383163899] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'IMU'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.383081083] [GenericAnalyzer]: Analyzer 'IMU' matches 'turtlebot4_diagnostics: /imu topic status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.381057909] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Camera'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.380978205] [GenericAnalyzer]: Analyzer 'Camera' matches 'turtlebot4_diagnostics: /color/preview/image topic status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.379311644] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Camera'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.379240962] [GenericAnalyzer]: Analyzer 'Camera' matches 'turtlebot4_diagnostics: /stereo/depth topic status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.377520431] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Lidar'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.377439838] [GenericAnalyzer]: Analyzer 'Lidar' matches 'turtlebot4_diagnostics: /scan topic status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.375495850] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.375473506] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Hazards'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.375396062] [GenericAnalyzer]: Analyzer 'Hazards' matches 'turtlebot4_diagnostics: Hazard Detections'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.374273257] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.374250931] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Dock'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.374166021] [GenericAnalyzer]: Analyzer 'Dock' matches 'turtlebot4_diagnostics: Dock Status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.372234076] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.372217401] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Wheels'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.372147850] [GenericAnalyzer]: Analyzer 'Wheels' matches 'turtlebot4_diagnostics: Wheel Status'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.371875592] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.371850525] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Battery'.
Feb 15 22:39:54 ubuntu turtlebot4-start[1359]: [aggregator_node-2] [INFO] [1676500794.371773433] [GenericAnalyzer]: Analyzer 'Battery' matches 'turtlebot4_diagnostics: Battery Voltage'.
[... cut ...]
Feb 15 22:39:50 ubuntu turtlebot4-start[1359]: [rplidar_composition-4] [INFO] [1676500790.086929116] [rplidar_composition]: RPLidar health status : '0'
Feb 15 22:39:50 ubuntu turtlebot4-start[1359]: [rplidar_composition-4] [INFO] [1676500790.085455319] [rplidar_composition]: Hardware Rev: 7
Feb 15 22:39:50 ubuntu turtlebot4-start[1359]: [rplidar_composition-4] [INFO] [1676500790.085413602] [rplidar_composition]: Firmware Ver: 1.29
Feb 15 22:39:50 ubuntu turtlebot4-start[1359]: [rplidar_composition-4] [INFO] [1676500790.085257165] [rplidar_composition]: RPLIDAR S/N: DB90EDF9C7E29BD1A7E39EF24B43431B
-- Reboot --

roni-kreinin, 2023-02-16:
The camera display is the correct one to use in RViz.

Can you try restarting the turtlebot4 service by calling `sudo systemctl restart turtlebot4` when the camera dies?

Fran-FC, 2023-05-16:
I use the latest packages and Ros Humble.

Hello, I don't know if it is a common issue but I have a problem with the stereo depth camera. I am able to visualize the rgb image of the oakd but not the stereo image. 

The journalctl output is the following:

```
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.868185396] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.835092079] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.779841089] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.734793267] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.686758694] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
May 16 10:48:00 ubuntu turtlebot4-start[2025]: [component_container-6] [ERROR] [1684234080.645433043] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
```

closed 2023-09-21 by hilary-luo