turtlebot/turtlebot4 #111: Unexpected Movement While Using Controller and When Charging In Dock

4DJR, 2023-02-14:
**Please provide the following information:**
Using Ubuntu 20.04 ROS2 Galactic Turtlebot4 Lite 0.1.3 Image

 **Expected behaviour**
When in dock: I expect the robot to stay in the dock while it is charging
When using PS4 controller: I expect the robot to only move when the joystick is moved

 **Actual behaviour**
When in dock: the Turtlebot will often undock itself before it is fully charged which causes the battery to die
When using PS4 controller: the Turtlebot will sometimes begin moving on its own and unexpectedly "do a little dance" when a user is driving it around. Note that the joystick controls were launched using "ros2 launch turtlebot4_bringup joy_teleop.launch.py"

**To Reproduce**
1) Put Turtlebot in dock to charge (without turning it off first)
2) Drive Turtlebot with PS4 controller (for ~15 mins)

Thanks.

roni-kreinin, 2023-02-14:
1. Is it performing the full "undock" action, or just moving slightly backwards?
2. Are you accidentally hitting the L2 or R2 buttons while driving? Those are mapped to the Wall follow left/right actions.

4DJR, 2023-02-14:
I'm told it is only moving slightly backwards. Also, I guess it was still doing the dance while the controller was resting on the table.

roni-kreinin, 2023-02-15:
If the bumper is depressed when the robot docks, any `cmd_vel` message will cause the bumper reflex to activate, making the robot drive backwards slightly. I have found that this will happen when closing `teleop_twist_keyboard`. I would recommend using the dock action rather than manually driving the robot onto the dock, if that is what you are doing. Otherwise, you can disable the bumper reflex. See https://iroboteducation.github.io/create3_docs/api/reflexes/

For the "dance", can you check the logs of the turtlebot4 service with `sudo journalctl -u turtlebot4 -r` to see if any of the Create 3 movement actions are being called? It sounds like something is calling these actions.

closed 2023-06-20 by roni-kreinin