turtlebot/turtlebot4 #113: Namespacing support for turtlebot4 teams

bbao200905, 2023-02-15:
We are interested in coordinating movements among turtlebots. For this, namespacing support would be necessary. The [user manual](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html#namespacing) states that this support will be coming soon. Is there a recommended approach for this?

roni-kreinin, 2023-02-15:
Namespacing support has been added to the `humble` branch. Because Galactic is EOL we aren't focusing on adding features to it any more. You can make a PR for it similar to the Humble [PR](https://github.com/turtlebot/turtlebot4/commit/b2f8f5c64fee5a616a9710718b4597cf8c126c09) if you are interested. Otherwise, I would recommend waiting for the Humble release.

Note that any updates to the galactic branch will not be released as new debian versions because Galactic is EOL.

roni-kreinin, 2023-02-15:
I don't have a specific release date but I am expecting to start releasing debian packages by next week. Once those are out, I will upload a new RPi image that you will be able to flash to your TB4. There will also be new Create 3 firmware to flash. Documentation updates will follow.

I would not recommend updating a TB4 running ubuntu 20.04 to the humble branch as there are some breaking changes.

Jayden-F, 2023-02-23:
Hello,  I am sorry I didn't see this earlier. After my conversation almost a month ago with Roni. I pulled the changes into a forked repo. Here are the branches for namespacing the robot. 

https://github.com/Jayden-F/turtlebot4_robot/tree/galactic-namespace
https://github.com/Jayden-F/turtlebot4/tree/galactic-namespace

To my knowledge the robot is successfully namespaced, however, I am struggling to run localization for multiple robots.

Hope it helps.

ipa-rar, 2023-02-27:
@Jayden-F Thank you for mentioning about your fork. I tried it for my use case and it did not work as expected. I could see the topics without any namespace. Can you provide some details on what you did to include namespace?

Jayden-F, 2023-02-27:
@ipa-rar I am assuming you have successfully uinstalled the binaries and built the packages from source. I have added a readme file to the forked repo. The readme file explains how to replace the robot upstart service.

roni-kreinin, 2023-03-06:
Humble has been released for the robot, including namespacing support. See #119 for details.

closed 2023-03-06 by roni-kreinin

sumitpaulde, 2023-12-07:
Hi, How can I read the Namespace value in the ROS2 node? Where is the value of the Namespace stored?