turtlebot/turtlebot4 #114: Cannot run Launch Files on Turtlebot4 Lite

rohanchandra-cpe, 2023-02-22:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed 
 - Package version: "sudo apt install ros-galactic-desktop" is how I installed ROS2

 **Expected behaviour**
Pressing Button1 on the Create3 produces a message in the terminal that says "Button 1 Pressed!" As per this link here: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_python.html

 **Actual behaviour**
No message appears upon pressing the button. Pressing CTRL+C to halt the command yields the error message attached in the image. 

**To Reproduce**
Provide the steps to reproduce:
1. Power on the Turtlebot4 (raspberry pi was previously set up to connect to a 5GHz network via wifi.sh)
2. Go to <ip of raspberrypi>:8080 and connect the Create3 to a 2.4GHz network
3. Follow https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_python.html until "Test Create3 Button1 "
4. Run "ros2 run turtlebot4_python_tutorials turtlebot4_first_python_node"
5. Press Button 1 on the Create3

**Other notes**
I am currently attempting to follow the guide for the Turtlebot 4 lite. Specifically the "Creating your first node (Python)" section: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_python.html. I get to the point where I need to run "ros2 run turtlebot4_python_tutorials turtlebot4_first_python_node." However, nothing happens when I press the first button on the Create3. Instead, when I CTRL+C out of the command, I get the error attached. I have connected the Raspberry Pi to a 5GHz network (wifi.sh) and the Create3 to a 2.4 GHz network. Please let me know how I can resolve this, Thanks!

[image]

roni-kreinin, 2023-02-22:
@rohanchandra-cpe Can you call `ros2 topic list` and paste the output here?

rohanchandra-cpe, 2023-02-22:
Sure thing. The first image is from my normal Ubuntu environment, and the second image was taken after ssh'ing into the raspberry pi. Should I be doing all of my development inside the Raspberry Pi?
[image]
[image]

roni-kreinin, 2023-02-22:
@rohanchandra-cpe It looks like there may be some networking issues going on.

Can you confirm that the Create 3 is connected to Wi-Fi by going to the webserver and checking the "Connect" tab? It should display the IP address if connected like this: 
[image]

If there is no IP address, try connecting the Create 3 again by following [these instructions](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#access-the-webserver-v012-or-higher).

It also looks like your PC cannot see the RPi topics. Can you make sure you are using the same `ROS_DOMAIN_ID` on both, and call `ros2 multicast receive` on your PC, followed by `ros2 multicast send` on the RPi to check that multicast packets aren't being blocked by your network?

Regarding development, you should be able to run the tutorial python node on either your PC or the RPi once all of the nodes can communicate with each other properly.

rohanchandra-cpe, 2023-02-24:
Ok, I ran the tests you wanted me to above:

1. An IP address does appear, so it looks like my Create 3 is connected
2. rosdomainids.png shows the result of searching for the ros domain ids. RPI on left, PC on right
3. the next image shows what happens when I did a ros2 multicast receive on my PC, followed by a ros2 multicast send on the RPI (RPI left, PC right). Nothing happened for about 15 seconds after I did the send command, so I did a CTRL+C on the right and I get that error message. 
[image]
[image]

roni-kreinin, 2023-02-24:
@rohanchandra-cpe Are you running the robot on a university or corporate network? It's possible that your network blocks multicast packets over Wi-Fi which is used for discovery.

If your network is blocking multicast packets, I would recommend following [these instructions](https://gist.github.com/roni-kreinin/8fbb20cb4603b8eb5e3961167fb22cd4) to configure your robot to use the FastDDS discovery server.

rohanchandra-cpe, 2023-02-24:
I am on my home network. We have a 5.0Ghz network and 2.4Ghz network. I have connected the former to the Raspberry Pi and the latter to the Create 3. According to the WiFi that I used to set up the RPi (wifi.sh), it should support multicast (image below). 

[image]

roni-kreinin, 2023-02-24:
Your Wi-Fi interface will support multicast, but your router may still block those packets. You can try going to your router gateway and looking for any multicast settings.

rohanchandra-cpe, 2023-02-24:
Ok, I'm going to try following the FastDDS instructions and see if that makes a difference. I'll let you know. Thanks!

rohanchandra-cpe, 2023-02-25:
The FastDDS instructions are not working for me. I get up to the point where I need to run 

sudo systemctl daemon-reload
sudo systemctl enable ip_route.service
sudo systemctl start ip_route.service

And I get the error below. 
[image]

I am running WSL 1.0.3.0.

roni-kreinin, 2023-02-27:
@rohanchandra-cpe Instead of running the `ip_route` service, you can manually add the ip route by calling `sudo ip route add 192.168.186.0/24 via xxx.xxx.xxx.xxx`, replacing the x's with your RPi IP address. You can add this line to your `~/.bashrc` file to have it run every time a new terminal is opened.

rohanchandra-cpe, 2023-02-27:
That did not work either. 
[image]
Is there any other information I can give you to help resolve this error?

rohanchandra-cpe, 2023-02-27:
Also, I already have that exact line you suggested I run in /usr/local/sbin/ip_route.sh on the remote PC. 

[image]

roni-kreinin, 2023-02-27:
@rohanchandra-cpe Can you paste the output of `ip a`?

rohanchandra-cpe, 2023-02-27:
There appears to be some sensitive information in that output. Is there anything specifically you want to see?

roni-kreinin, 2023-02-27:
Your Wi-Fi interface. You may need to call something like `sudo ip route add 192.168.186.0/24 via 192.168.1.171 dev wlan0`, replacing `wlan0` with your interface name.

rohanchandra-cpe, 2023-02-27:
This is what I currently see after starting a fresh new terminal
[image]

roni-kreinin, 2023-02-27:
Can you make sure that you can ping the Create 3 from your PC by calling `ping 192.168.186.2`? Also try calling `ros2 daemon stop` before calling `ros2 topic list`. You may need to call `ros2 topic list` a couple of times to get the topics to show up. Also, SSH into the RPi and check that you can list the topics there.

Lightring states are shown [here](https://iroboteducation.github.io/create3_docs/hw/face/).

rohanchandra-cpe, 2023-02-27:
I am able to ping that IP address. I also did a ros2 daemon stop before calling ros2 topic list. However, nothing has appeared in the ros2 topic list output in neither the RPi nor my remote PC.

rohanchandra-cpe, 2023-02-27:
By the way, I do not see an IP address in the online portal for the Create 3. Is this an issue, or to be expected from a FastDDS setup?

[image]

rohanchandra-cpe, 2023-02-27:
[image]

[image]

[image]

roni-kreinin, 2023-02-28:
The `systemctl` status is showing a time jump from when the services were started. When you call `date` on your robot, does it display the correct time and date?

rohanchandra-cpe, 2023-03-05:
Hey, sorry for the delay @roni-kreinin . Here are the screenshots you asked for.

date appears to show the correct time and date:
[image]

And here is htop on the robot:
[image]

rohanchandra-cpe, 2023-03-08:
@roni-kreinin I am still struggling with this issue. When you get a chance please see the information I attached above. Thank you

roni-kreinin, 2023-03-08:
Can you try stopping the turtlebot4 service with `sudo systemctl stop turtlebot4`, then restarting the discovery server service with `sudo systemctl restart discovery`. Once it has restarted, manually launch the turtlebot4 nodes with `ros2 launch turtlebot4_bringup lite.launch.py`. Then, open a new terminal in the RPi and run `ros2 daemon stop`, then `ros2 topic list`. You may need to run `ros2 topic list` twice to get a full list of available topics.

rohanchandra-cpe, 2023-03-08:
[image]

I did everything you asked for. The screenshot is from the second Rpi terminal you asked me to make. It looks like I got all the topics at first, but then with multiple "ros2 topic list" calls it went away

roni-kreinin, 2023-03-09:
@rohanchandra-cpe There is likely some configuration that is missing, though I am not seeing anything wrong. Would you be interested in updating to Humble? There is a new setup tool that takes care of all this configuration automatically.

See https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#robot

roni-kreinin, 2023-03-22:
@rohanchandra-cpe Did you update the RPi4 to Ubuntu 22.04 manually? Or did you use a humble image from http://download.ros.org/downloads/turtlebot4/?

roni-kreinin, 2023-03-22:
Can you try to follow https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image to install the official humble image?

rohanchandra-cpe, 2023-03-22:
Will I need to ssh into the rpi and do this, because if so, I currently cannot ssh into it. The old IP address I used to ssh to does not work anymore. And if I try to ping it, I just get "Destination Host Unreachable"

roni-kreinin, 2023-03-22:
No, you will have to remove the SD card from the RPi and flash it with a new image. Note that this will overwrite all data on the SD card, so save any changes you have made before doing so.

rohanchandra-cpe, 2023-03-22:
Ok, I'll do that and let you know how it goes. Thanks!

rohanchandra-cpe, 2023-03-22:
[image]

It said there wasn't enough space on the microsd for the pi. I noticed a lot of these files above that all have the same date (3/20/23), which is when I did the manual upgrade. Can I safely delete these?

roni-kreinin, 2023-03-22:
Where are you seeing this error?

rohanchandra-cpe, 2023-03-22:
Ah yes, sorry. I do

roni-kreinin, 2023-03-22:
Is this running on a VM?

rohanchandra-cpe, 2023-03-22:
This is what I see on the Powershell: The D: is my SD Card
[image]

roni-kreinin, 2023-03-22:
I'm not too familiar with WSL but you can try following a tutorial like this to mount the SD card. https://learn.microsoft.com/en-us/windows/wsl/wsl2-mount-disk

roni-kreinin, 2023-03-23:
Run the suggested `ssh-keygen` command in the warning message, then try to ssh again.

roni-kreinin, 2023-03-23:
>The top window is what happened after I finished running the setup tool. I think the RPI rebooted twice, but that top terminal has not changed at all. The bottom terminal shows what happened after I tried calling "ros2 topic echo /ip"

The ssh session will freeze because the RPi has changed networks. You can close that terminal.

rohanchandra-cpe, 2023-03-23:
Ok, I was able to ssh into it. What should I do from here?

roni-kreinin, 2023-03-23:
Seems to be missing the Create 3 topics. Did you update the Create3 to the latest Humble firmware? https://iroboteducation.github.io/create3_docs/releases/h_1_1/

rohanchandra-cpe, 2023-03-23:
[image]

I think so

rohanchandra-cpe, 2023-03-23:
Just one

rohanchandra-cpe, 2023-03-23:
I followed the above link up until User PC (I have not done the User PC steps yet). This is what I see when I call "ros2 topic list"
[image]

rohanchandra-cpe, 2023-03-23:
[image]

I started the User PC config, I got this error

rohanchandra-cpe, 2023-03-23:
[image]
Some more testing I did (ip route did have the additional ip address)

rohanchandra-cpe, 2023-03-23:
Also @roni-kreinin I selected the defaults for the Discovery Server IP address and port

roni-kreinin, 2023-03-23:
The `ros2 topic list` on the RPi looks correct.

I think WSL does not use `systemd` so the ip_route service won't work. You will have to add the `sudo ip route add ...` command to your `~/.bashrc` file to have it executed every time you open a terminal. If `ip route` is showing the route then you should be able to ping the Create 3 from your PC by calling `ping 192.168.168.2`.

If you are able to ping then you can check if the ros2 topics are showing up by calling 
```
ros2 daemon stop; ros2 daemon start
ros2 topic list
```

rohanchandra-cpe, 2023-03-23:
Ok, adding the stuff at the end did work. This is now what I see when I do "ip route"
[image]

But I cannot ping 192.168.168.2
[image]

roni-kreinin, 2023-03-23:
Does your Wi-fi interface show up in WSL?

rohanchandra-cpe, 2023-03-23:
How would I check that?

rohanchandra-cpe, 2023-03-24:
@roni-kreinin Is this what I would need to check to see if my wi-fi interface is in WSL?

roni-kreinin, 2023-03-24:
You might need to do something like [this](https://learn.microsoft.com/en-us/windows/wsl/networking#accessing-a-wsl-2-distribution-from-your-local-area-network-lan) to get WSL on the same network as your Windows PC. Or you can try running a VM instead.

rohanchandra-cpe, 2023-03-24:
Honestly, I think at this point a Linux VM would be the best option. If I were to go that route, I imagine I would have to do the entire User PC configuration again, but would I need to do any of the Raspberry PI/Create3 configs again?

roni-kreinin, 2023-03-24:
You would just need to do the PC configuration.

rohanchandra-cpe, 2023-03-26:
@roni-kreinin I have lost the ability to ssh and ping the raspberry pi. 
[image]

As you can see I am now in rohan@asterix, which is a linux vm running Ubuntu 22.04 LTS. I am no longer using WSL.

When I went through the discovery server steps on this VM, everything worked just fine, including "ip route"

[image]

rohanchandra-cpe, 2023-03-27:
@roni-kreinin I tried again today, still cannot ssh in. Could this be because of the port number?

rohanchandra-cpe, 2023-03-27:
@roni-kreinin VMWare EXSI 6.5.0 (VMKernel Release Build 4564106)

rohanchandra-cpe, 2023-03-28:
@roni-kreinin How would the VM I am using affect my ability to SSH into the RPi?

roni-kreinin, 2023-03-28:
You may need to do some configuration to bridge the VM network to your PC. Try this https://www.minitool.com/backup-tips/vmware-bridged-network-not-working.html

[3 more replies]

closed 2023-04-29 by rohanchandra-cpe