turtlebot/turtlebot4 #116: Communication LED is off

languyen14507, 2023-02-25:
Hello,
I am having another problem with my Turtlebot 4. For some reason, the onboard display where I used to see the bot's IP address and undock, dock,.. selections is completely off. I can connect to the Create@3 via the Create-xxxx net work and verify that the Create@3 is publishing its topics
...
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

but I can't drive the robot because I can't undock it. I tried to also use the teleop_twist_keyboard node but it doesn't work either. Please advise. Thank you.

languyen14507, 2023-02-25:
Please note also that I can no longer ssh to Turtlebot 4 (ssh ubuntu@192.168.11.134)

roni-kreinin, 2023-02-27:
@languyen14507 It sounds like the RPi isn't able to connect to Wi-Fi for some reason. Has your network changed?

To recover the RPi, you can SSH into it via an Ethernet cable. See https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-rpi-over-ethernet. Once you are SSH'd in, call `ip a` to check the status of the `wlan0` interface. If your Wi-Fi network has changed, call `wifi.sh` to reconnect the Pi with your new credentials.

languyen14507, 2023-02-28:
I have bought an ethernet cable to connect my Ubuntu 20.04 to the Rpi. How do I configure my wired connection to use the same subnet as the RPi? I don't know what my static IP address is.

languyen14507, 2023-02-28:
My Wi-Fi network has not changed.

languyen14507, 2023-02-28:
[image]

tonybaltovski, 2023-02-28:
Set the static IP as shown here: https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/media/static_ip.png

languyen14507, 2023-02-28:
I did that (using the IP addresses 192.168.185.5, 255.255.255.0, 192.168.185.1) but when I tried to ssh ubuntu@192.168.185.3, I got a connection refused response.

tonybaltovski, 2023-02-28:
Just to clarify, your image showed those fields as empty.    What is the output of `ip a` on your machine?

languyen14507, 2023-02-28:
I did have the correct numbers in those fields and hit "Apply", the image I took was before I did that. However, I just powered both turtlebot 4 and my PC then powered them back up and miraculously, I can now talk to the RPi and Create@3. The LED display is back on except for some reason, when I push the "home" button on the hand controller, it does flash but does not turn solid blue and just keep flashing, so I cannot undock the bot.

roni-kreinin, 2023-02-28:
@languyen14507 You can try to follow the [controller setup](https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html#turtlebot-4-controller-setup) to reconnect the controller

closed 2023-03-01 by roni-kreinin