turtlebot/turtlebot4 #117: Tf has two or more unconnected trees.

Basavaraj-PN, 2023-02-28:
**Sys information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source
 - Package version: 
 - create3 version: 4.3 on 2.4Ghz

 **Expected behaviour**
 Should launch nav2 core and rviz should show map.

 **Actual behaviour**
Timed out waiting for transform from base_link to odom to become available, tf error: Could not find a connection between 'odom' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.

**To Reproduce**
Provide the steps to reproduce:
1. ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=async map:=~/map_name.yaml
2. 
3. see the termianl output

roni-kreinin, 2023-02-28:
This is likely related to the errors you were seeing [here](https://github.com/turtlebot/turtlebot4/issues/62#issuecomment-1447572362). I have asked iRobot for insight on these errors.

roni-kreinin, 2023-04-17:
@Basavaraj-PN Sorry for the delay. 

I have a few questions:
- Can you paste the output of `ros2 topic list` from the RPi and your PC?
- What does your network look like? Are you using a router for your Wi-Fi or a mobile hotspot?
- Have you tried using the discovery server as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)?

From the errors in the other issue, it seems that your Create 3 is having issues connecting to your Wi-Fi network, so I am assuming that you are not seeing topics from the Create 3.

Basavaraj-PN, 2023-04-25:
@roni-kreinin 
On my PC
/battery_state
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/camera_info
/stereo/depth
/tf
/tf_static
/wheel_status

/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/dock_status
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

On RPi

Basavaraj-PN, 2023-04-27:
I tried that, the main issue is ntpd, time synchronisation issue

Basavaraj-PN, 2023-04-28:
Even if I want to run everything in RPi things are failing because of time synchronisation issue. 
https://github.com/turtlebot/turtlebot4/issues/10

roni-kreinin, 2023-04-28:
Can you show me what errors you are getting with regards to time sync?

Basavaraj-PN, 2023-05-15:
https://github.com/turtlebot/turtlebot4/issues/10 @roni-kreinin

roni-kreinin, 2023-06-19:
@Basavaraj-PN Can you share what your Wi-Fi setup looks like, and the actual sync errors that you are getting?

Basavaraj-PN, 2023-06-22:
Hi @roni-kreinin , we are installing Ubunut 22.04 in RPi and humble image in create.
We will share the update to you after trying it out with latest binaries.

closed 2023-09-21 by hilary-luo