turtlebot/turtlebot4 #120: Flashing MicroSD to humble

Jayden-F, 2023-03-06:
**Please provide the following information:**
 - Flashing turtlebot4 lite with turtlebot4_lite_humble_1.0.0.img

 **Expected behaviour**
- Raspberry pi starts in AP mode 
- Accessible over eth via 192.168.185.3

 **Actual behaviour**
- Solid Red Power Light on rpi.
- Solid Green ack Light on rpi.

- Not in AP mode.
- Not Accessible over eth.

**To Reproduce**
Provide the steps to reproduce:
1. flash micro sd using steps from documentation. (https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image)
2. Insert micro sd card
3. Turn on turtlebot4 
4. No AP or eth access.

roni-kreinin, 2023-03-06:
@Jayden-F Can you confirm that the file transfer finished successfully? It should look like this:
```
[97% of 7365Mb] 7168 blocks (7168Mb) written. 00:00:05 remaining.
7365+1 records in
7365+1 records out
```

Jayden-F, 2023-03-06:
@roni-kreinin 
Hello Roni, Yes I can see the transfer has finished successfully.
~~~
Image path: turtlebot4_lite_humble_1.0.0.img
Enter each SD card device name separated with a space (i.e. sda sdb sdc): sdc1
The SD card(s) will be unmounted and flashed. Press enter to continue.
umount: /dev/sdc1: not mounted.
[97% of 7365Mb] 7168 blocks (7168Mb) written. 00:00:02 remaining.
7365+1 records in
7365+1 records out
~~~

roni-kreinin, 2023-03-06:
@Jayden-F Can you try running it again using the device `sdc` rather than `sdc1`? I believe `sdc1` is just a partition of the SD card.

Jayden-F, 2023-03-06:
Thank you, That was it!

closed 2023-03-06 by Jayden-F