turtlebot/turtlebot4 #121: Cannot power on turtlebot4

VincentV89, 2023-03-07:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**
 A clear and concise description of what you expected to happen.
Power on at dock

 **Actual behaviour**
 A clear and concise description of what you encountered.
Not powering on when connected to dock. Dock have pulsing yellow light

**To Reproduce**
Provide the steps to reproduce:
1. connect to dock

**Other notes**
Add anything else you thing is important.

roni-kreinin, 2023-03-09:
@VincentV89 Can you remove the battery from the robot, wait 15-20 mins, then place it back in? You will have to remove the bottom panel on the robot which is secured by 5 screws to access the battery.

JHamlin244, 2023-04-28:
I am having a similar issue. Robot does not power on at all. 
I have unplugged the battery overnight, there is no change when trying to boot the robot.

There is no light in the button ring, but the "internal 5V" light on the adapter board is flashing.
https://iroboteducation.github.io/create3_docs/hw/adapter/ 

Attached video 

https://user-images.githubusercontent.com/132094793/235221215-882308fb-5dac-40fe-b02d-324dd0d174a1.MOV

This occured after being powered off for about a week.

JHamlin244, 2023-05-02:
I have purchased a second battery, in case that was the issue. There is no change in behavior. It is still flashing and will not power on..

JHamlin244, 2023-05-04:
@roni-kreinin are you, or anyone else, able to help with this item?

closed 2023-09-21 by hilary-luo