turtlebot/turtlebot4 #124: Battery Management Issues in ROS Humble RPI Image

BashByr, 2023-03-17:
**Please provide the following information:**
 - OS: Ubuntu 22.04.2 LTS
 - ROS Distro: (e.g. Galactic): Humble
 - Built from source or installed: Installed
 - Package version: “turtlebot4_standard_humble_1.0.0.zip | 2023-03-03 23:51 | 1.9G” from http://download.ros.org/downloads/turtlebot4/

 **Expected behaviour**
The Turtlebots (both Lite and Standard) should realize that they are fully charged and not power off randomly, or realize that they are not charged and should charge up when docked. 

**Actual behaviour**
Instead, when I place them on their docks, they first boot up normally and publish their topics. However, after a few minutes (and at exactly the same time if they were both initially docked at the same time), they turn off, and their light rings turn red (the red LEDs fill up the circle one by one and the pattern loops). Interestingly, they both seem to still be publishing their topics as seen below:

[image]

Looking at the /battery_state topic of the TurtleBot4 Standard:

[image]

This was quite strange, as just a few minutes before, I had connected the controller to the bot and drove it off its dock. When I did that and looked at its /battery_state, I got this:

[image]

Indicating a 99% state of charge. Even the LED screen echoed the same thing. However, shortly after that was published, the robot turned off. I then placed it back on its dock, where it booted back up and then displayed the red ring and shut off a few minutes later. The /battery_state topic reverted to publishing a 0% SOC. It seems to be some kind of standby mode, as they still publish their topics, but they immediately die if I lift them off their docks. I confirmed this by subscribing to one of their /tf topics and seeing a stream of messages. So they recognize that they are being charged, but do not seem to charge. Also, the ring light on the Lite will sometimes turn a solid yellow:

[image]

Currently, they are both docked, with the Lite displaying the yellow ring above and the Standard displaying the red animated ring. I keep checking their battery states but they are both at 0% and have been for some time, despite them being docked. As I mentioned, if I lift them off their docks, they immediately and completely shut off.

**To Reproduce**
The robots just power off and display their red light rings a few minutes after they’ve been docked.

**Other notes**
Add anything else you thing is important.

BashByr, 2023-03-17:
Update: shortly after this post, I undocked and redocked the TBot4 Lite and it went back to the red LED animation. Both the Lite and Standard have been docked for around 6 hours and still display the red LED animation.

roni-kreinin, 2023-03-17:
@BashByr Can you upload the logs from the Create 3 webserver for both robots after this happens? Can you also check the battery levels in the  "About" section? There should be an entry like this:

[image]

The `turtlebot4_node` has logic in it to automatically power the robot off if it goes below 12% charge for over a minute, and because it is receiving those 0's in the battery topic, I suspect that this logic is being triggered.

BashByr, 2023-03-22:
@roni-kreinin I really appreciate your quick responses. I am not able to provide logs _after_ the robots shut down and display the red ring light as the Create 3 webserver can no longer be reached at that point. However, I have logs shortly before they shut down, as described below:

For the TBot4 Lite, I first undocked it to completely shut it down, then redocked and had it boot back up. Once it booted up, there is a brief period during which I can access the Create 3 webserver before the robot goes into low power mode. Here, I got the following from the About section, indicating a 100% SOC:

[image]

Strangely enough, looking at the /battery_state topic for that robot, I kept getting a 0% SOC and it wouldn’t even recognize it is docked and charging:

[image]

Here is the Create 3 webserver log shortly before it shut off into low power mode:

<details>
<summary> TBot 4 Lite Create 3 Log </summary>
`Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 syslog.info syslogd started: BusyBox v1.30.1
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.notice kernel: klogd started: BusyBox v1.30.1 (2022-09-13 23:06:29 UTC)
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Tue Sep 13 23:16:14 UTC 2022
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Board has DDR2
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Analog PMU set to hw control
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Digital PMU set to hw control
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Determined physical RAM map:
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82d9f000 (2059380 bytes)
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Zone ranges:
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Movable zone start for each node
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Early memory node ranges
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80562820, node_mem_map 81000060
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Mar 22 03:00:16 iRobot-04164D8E666A4719B91EFDEAABB109D1 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[... cut ...] t
Mar 22 03:40:08 iRobot-8748D7279D6349E98FCA76C83EF27D99 user.notice create-platform: VSTAT: No config for vstat_nav_os_temp_c
Mar 22 03:40:08 iRobot-8748D7279D6349E98FCA76C83EF27D99 user.notice create-platform: VSTAT: No config for os_cpu_active
Mar 22 03:40:08 iRobot-8748D7279D6349E98FCA76C83EF27D99 user.notice create-platform: VSTAT: No config for vstat_nav_os_meminfo_kb
Mar 22 03:40:08 iRobot-8748D7279D6349E98FCA76C83EF27D99 user.notice create-platform: [INFO] [1679456408.776698049] [tbot1.

BashByr, 2023-03-22:
Both of them are are running G.3.1:

TBot Standard:
[image]

TBot Lite:

[image]

BashByr, 2023-03-22:
Hadn’t even considered that. Just did, and they’re both back to normal now. Thanks a ton.

closed 2023-03-22 by BashByr

schenkch, 2023-05-17:
@roni-kreinin I have a similar issue with the turtlebot4 standard but in my case the battery is at 15%. The lightring is red and when I put it on the dock, it charges for like two minutes (the color of the lightring changes back to normal color) and then it turns red again and the green light on the docking station turns off. I checked whether it is a cable problem but it does not seem to be. Any idea on how to resolve this?

schenkch, 2023-05-17:
I am having a related problem with Galactic and the turtlebot4 standard. I was running ros2 launch turtlebot4_navigation nav_bringup.launch.py localization:=true slam:=async map:=map_office.yaml, ros2 launch turtlebot_viz view_robot.launch.py and modified the script from the tutorial for multiple sequential goals instead of just one. I checked the battery state before navigation which was around 60% and while navigating to the second goal the robot just stops, the lightring turns red and the battery state went down to 16%. It is consuming a lot of battery. Is there a fix for this? @roni-kreinin Or would this be resolved with upgrading from the latest Galactic to the latest Humble version?