turtlebot/turtlebot4 #125: Many problems

MathisGodard, 2023-03-21:
Hello, our customer has some problems with his turtlebot 4.
Find below the configuration of the robot : 

OS: Ubuntu 20.04.5
ROS Distro: Galactic
Installation/Build: Following the instructions that were present under “https://turtlebot.github.io/turtlebot4-user-manual/overview/quick_start.html”, and are now similar under “https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html”. For ROS2 install, the linked instructions under “https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html” were used.

Package version:
                Ros-galactic-desktop: 0.9.3-2focal.20221208.082310
                Ros-galactic-ros-base: 0.9.3-2focal.20221208.073437
                Ros-dev-tools: 1.0.0

1. Indicator lamps are not lit
When turning on the robot, the indicator lights on top of the robot sometimes light up not, even if it should work.

2. Display remains black
The display often simply remains black after turning on the robot, which not only makes it makes it difficult to navigate the standard programs of the robot (e.g. auto-dock), but also makes it difficult to determine the IP address of the robot

3. The Ros2 topic /ip should be shared by the robot by default, but it will not

4. The joy_teleop-nodes should run on the robot by default, but they don't always.

5. When the robot is running, it responds to /cmd_vel commands, but not to the Ros2
Actions of Create3 (not extensively tested)

6. The robot does not have the same IP address on the same network
Unfortunately, the robot in our network often changes its IP address by switching it off and on. (Could be how our network is set up)

7. Sometimes the joy_teleop-nodes cannot even be started successfully manually

8. When the robot is moving, after a few minutes of driving and/or a few meters
Removal of charging station and router one (RANDOM) of its indicator lights off and he stops responding to the controller

Best regards, Mathis from Génération Robots.

roni-kreinin, 2023-03-21:
It sounds like there are some networking issues going on. Is it possible that the Wi-Fi connection is weak where the robot is being tested?

The LEDs and display being off indicates that the turtlebot4 node has crashed in the background. This only really happens with cycloneDDS when one of the network interfaces has disconnected. Can you ask the user to SSH into the robot and check the output of `sudo journalctl -u turtlebot4 -r` for any errors?

The user can set a static IP address by modifying their `/etc/netplan/50-cloud-init.yaml` file. They can add the static address under the `wlan0` interface like this:
```
wifis:
    wlan0:
      addresses:
        - 10.10.10.2/24
```

It also sounds like for number 7 and 8 the Create 3 is disconnecting from Wi-Fi and so the COMMS LED is turning off, and the robot cannot drive.

My recommendations would be:
- Update the robot to Humble to get the latest features and the [turtlebot4_setup tool](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools). This will make configuring things like the static IP much simpler.
- Switch to FastDDS to avoid ROS2 crashing when one of the interfaces fails
- Bring the robot closer to the router or upgrade the router.
- Switch to using the [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) to use only the RPi4 Wi-Fi connection.

MathisGodard, 2023-04-17:
Hello,

Find attached the message from the customer.  

We tried to follow the recommendations to the best of our ability and are still having issues (possibly even bigger issues):
The result from the journal command mentioned by the manufacturer can be found in the attached file “turtlebot4JournalResults”.
After that we tried to take the steps mentioned in the attached file “turtlebot4TakenSteps”.
After these steps we are now no longer able to ssh to the robots’ raspberry pi or connect to it with an ethernet cable.

Best regards, Mathis

[turtlebot4JournalResults.pdf](https://github.com/turtlebot/turtlebot4/files/11249952/turtlebot4JournalResults.pdf)
[turtlebot4StepsTaken.pdf](https://github.com/turtlebot/turtlebot4/files/11249954/turtlebot4StepsTaken.pdf)

roni-kreinin, 2023-04-17:
Any commands that return "Permission denied" or "Access Denied" should be called as `sudo`.

To update the RPi they should reflash the SD card by following these instructions: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

MathisGodard, 2023-04-25:
Hello, 

the customer ran into some unexpected errors again.

-downloaded standard humble zip from  http://download.ros.org/downloads/turtlebot4/
-unpacked zip into my user directory 
-removed microSD from Robot Raspberry Pi
-connected microSD to my Laptop as instructed
-installed dcfldd
-found SD-card name "sda"
-got sd flash-script as instructed
-ran bash sd_flash.sh /home/MYUSER
	-followed instructions from script
	-sucessfully flahed sd card
-removed sd from PC
-inserted flashed SD card to Robots Raspberry Pi
-Starting to follow Robot setup:
	-display did not turn on again
	-wifi does not seem to be reset
	-getting "command not found" error on trying to run "turtlebot4-setup"

Thanks, Best regards

MathisGodard, 2023-06-05:
Hello, some news regarding this case : 

Sadly, we are still struggling with getting our turtlebot4 up to working standards.

We tried installing the newest raspberry pi image with ros2_humble on it again, but it seems like we were unsuccessful again.

As always, additional information is provided in the attached file(s) and we included a picture of the result of running the flash program for 1 hour.

[image]
[turtlebot4TakenSteps.txt](https://github.com/turtlebot/turtlebot4/files/11652833/turtlebot4TakenSteps.txt)
[turtlebot4StepsTaken.pdf](https://github.com/turtlebot/turtlebot4/files/11652836/turtlebot4StepsTaken.pdf)

Best regadrs, Mathis.

roni-kreinin, 2023-06-06:
@MathisGodard That does look like the correct output from the `sd_flash.sh` script, however it usually takes ~5 minutes to flash, not an hour. If the user has a micro HDMI cable they can try to connect the RPi4 to a monitor with it to see if it is booting. If not, we can arrange for a new SD card to be sent to the user.

MathisGodard, 2023-07-17:
Hello, 

thanks for the additional guidance. My colleague has just now confirmed all of the points you specified: create3 is booting properly and the RasPi is turned on. We re-imaged the sd-card after removing the pcba to access the RasPi. The issues remain.

So, in case we still expect an issue with the re-imaging of the SD-card, we would appreciate receiving a new one. Shipping address would be the following:

Mathis Godard - Génération Robots
10 rue joule
33700 Merignac 
France

Best regards, Mathis.

roni-kreinin, 2023-07-19:
@MathisGodard An SD card is being sent out to you.

closed 2023-09-21 by hilary-luo