turtlebot/turtlebot4 #128: Map Server Error

Jayden-F, 2023-03-22:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source
 - Package version: Turtlebot4: df1aff80e816dee72758616b56882b643a2d761e

 **Expected behaviour**
run the following command: ros2 launch turtlebot4_navigation localization.launch.py
map server and localization server start up

 **Actual behaviour**
Error and the following log
[log.txt](https://github.com/turtlebot/turtlebot4/files/11045147/log.txt)

**To Reproduce**
Provide the steps to reproduce:
1. run the following command: ros2 launch turtlebot4_navigation localization.launch.py

Jayden-F, 2023-03-23:
Hello Roni, Yes, it been consistent since changing to Humble and using the fast dds server setup.

Here is another log from today:
[log2.txt](https://github.com/turtlebot/turtlebot4/files/11056997/log2.txt)

roni-kreinin, 2023-04-14:
@Jayden-F  Are you still encountering this issue?

I have been able to reproduce it with a newly generated map. Interestingly I seem to have fixed it by changing the `free_thresh` parameter to 0.196 in the map .yaml file. I'm not sure why this resolved the issue, but you can try that to see if it fixes the problem.

closed 2023-06-14 by Jayden-F