turtlebot/turtlebot4 #129: Multiple Robots (namespaced) - create3 error state

MatteoCarlone, 2023-03-29:
**Please provide the following information:**
 - OS:  Ubuntu 22.04
 - ROS Distro: Humble

We updated the turtlebot and create3 to humble in order to try the namespacing. In the beginning we had troubles with the default communication (FastRTPS) and we switched to CYCLONEDDS. 
We are testing everything on two T4 and we can see the robots' topics correctly namespaced, but most of the time one of the create3 run into error state. Sometimes the create can recover by keeping pressed the power button sometimes it start to boot forever.
Is this a known issue and is it already clever how to fix?

roni-kreinin, 2023-03-29:
Can you get the logs from the Create 3 webserver when it runs into this issue?

etigafy, 2023-03-31:
Thank's for your help, there are the Log's from the Create:

```
Mar 31 09:00:07 iRobot-3 syslog.info syslogd started: BusyBox v1.30.1
Mar 31 09:00:07 iRobot-3 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Tue Feb 14 17:10:20 UTC 2023
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Board has DDR2
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Analog PMU set to hw control
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Digital PMU set to hw control
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Determined physical RAM map:
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82d9f000 (2058981 bytes)
Mar 31 09:00:07 iRobot-3 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Mar 31 09:00:07 iRobot-3 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Zone ranges:
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Movable zone start for each node
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Early memory node ranges
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80562820, node_mem_map 81000060
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Mar 31 09:00:07 iRobot-3 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Mar 31 09:00:07 iRobot-3 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Mar 31 09:00:07 iRobot-3 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Writing ErrCtl register=0006ffb5
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Readback ErrCtl register=0006ffb5
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] Memory: 56900K/65536K available (4211K kernel code, 192K rwdata, 1112K rodata, 172K init, 218K bss, 8636K reserved, 0K cma-reserved)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] NR_IRQS: 256
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] intc: using register map from devicetree
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] CPU Clock: 575MHz
Mar 31 09:00:07 iRobot-3 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6647862422 ns
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000012] sched_clock: 32 bits at 287MHz, resolution 3ns, wraps every 7469508094ns
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.000057] Calibrating delay loop... 380.92 BogoMIPS (lpj=1904640)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.059947] pid_max: default: 32768 minimum: 301
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.060232] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.060248] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.062625] devtmpfs: initialized
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.067447] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Mar 31 09:00:07 iRobot-3 kern.info kernel: [    0.067469] futex hash table entries: 256 (order: -1, 3072 bytes)
[... cut ...]
Mar 31 09:53:37 iRobot-3 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 67s
Mar 31 09:53:50 iRobot-3 user.notice create-platform: [INFO] [1680256430.242812005] [robot3.system_health]: CPU usage: max 97 [%] mean 70 [%] RAM usage: 33/59 [MB]
Mar 31 09:54:50 iRobot-3 user.notice create-platform: [INFO] [1680256490.242972026] [robot3.system_health]: CPU usage: max 100 [%] mean 70 [%] RAM usage: 33/59 [MB]
Mar 31 09:55:01 iRobot-3 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 66s
```

roni-kreinin, 2023-04-05:
@MatteoCarlone Do you have any issues with running a single namespaced robot?

etigafy, 2023-04-13:
Sorry for the late answer, yes we also have troubles with a single namespaced robot. These are the error logs from today:

```Apr 13 08:26:58 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/aos_net.c:449 errno:Address not available) route does not belong to main routing table
Apr 13 08:26:58 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/aos_net.c:249 errno:Address not available) aos_net_parse_routes failed
Apr 13 08:26:58 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/aos_net.c:449 errno:Address not available) route does not belong to main routing table
Apr 13 08:26:58 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/aos_net.c:249 errno:Address not available) aos_net_parse_routes failed
Apr 13 08:27:04 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/mqtt_client.c:695) firr_transfer failed
Apr 13 08:27:04 iRobot-05E98F65C6014546B5B68C1021176AE0 daemon.err network[1468]: [ERROR] (src/discovery.c:407) falied to get cap from tsprime
```

The odom is not visible as well as the battery_state is not publishing, so we can see the battery_state topic but ros2 topic info says that it is not published. The `COMM` as well as `BAT` Led is off.

BashByr, 2023-04-24:
@roni-kreinin adding to this, this issue is exactly what I described in #132. Each namespaced robot has no issue working on its own. However, once the second robots turns on, one of the robots always errors out in a few minutes. Looking at the Create3 logs (in the brief time that I am able to do so before losing access to the webserver), I can see that the CPU maxes out to 100%. With just one robot on, this error never happens.

closed 2023-09-21 by hilary-luo