turtlebot/turtlebot4 #132: TurtleBot Topics Still Interfering with Each Other After Namespacing

BashByr, 2023-04-04:
**Please provide the following information:**
 - OS: Ubuntu 22.04.1
 - ROS Distro: Humble
 - Built from source or installed: Source

 **Expected behaviour**
When two namespaced TurtleBots connect to the same WiFi, I’d expect them to publish all their topics in the same way they did when they were individually connected with the other one turned off.

 **Actual behaviour**
I have confirmed this interference by first booting up one robot, waiting for it to fully connect to WiFi and publish everything, then turn on the second robot. The second robot usually has a hard time connecting to wifi, and usually does not publish all its topics no matter how many times I restart the system using turtlebot4-service-restart (the OAKD topics being one example). Only one robot typically publishes OAKD topics at a time, even after namespacing them. Another instance of this happens where, instead of the second robot not connecting, it successfully connects but the first robot reverts to a red or yellow light ring, and never properly reconnects.

To enable namespacing, did I only have to do that in turtlebot4-setup on the RPI? I suspect something is left without a namespace and begins interfering whenever two robots are connected. As I mentioned, what usually happens is one connects, and the other displays a solid red light ring. I have had great difficulty shutting off a TurtleBot after the solid red light ring is displayed, as it no longer will respond to pressing and holding the power off button to shut it down. Also, since I’m assuming solid red means a wifi/communication issue, I am no longer able to ssh into the RPI to restart anything. When this happens, I typically just leave the robot undocked until it runs out of battery and shuts off. Is there a known way to shut off/reboot a TurtleBot in this state?

**To Reproduce**
1. Have two namespaced TurtleBots and try to display each of their OAKD topics simultaneously.

**Other notes**
Is there anything else I need to change in the bottom two screenshots?

[image]

[image]

I tried looking at the “status” tab in turtlebot4-setup while SSH’d into the RPI to see why one tbot (tbot1) published its OAKD topics but the other (tbot2) did not:

[image]

[image]

For some reason, tbot2 says “OAKD stopped” and “RPLIDAR stopped” despite restarting the turtlebot4 services. These issues completely resolve when only one TurtleBot is on at a time. Could this be a problem between Create3 and my router?

roni-kreinin, 2023-04-04:
@BashByr The OAKD stopped and RPLIDAR stopped messages are from the power saving mode enabled [here](https://github.com/turtlebot/turtlebot4_robot/blob/3d92a3f2b7ea889e198352322e231a7fab540dcf/turtlebot4_bringup/config/turtlebot4.yaml#L8). This mode calls the rplidar and oakd stop services whenever the robot is docked, and starts them back up when the robot undocks. You are probably not seeing OAK-D data on docked robots because of this.

Regarding red lightring issues and the Create 3 being unresponsive, there are currently some performance issues with FastDDS on the Create 3 when not using the discovery server. The CPU hits 100% and the robot becomes unresponsive. iRobot is looking into it but for now I would recommend switching to CycloneDDS.

If your robot gets into an error state where it won't turn off, press and hold the power button for 20-30s. This will cause a hardware reset of the Create 3.

BashByr, 2023-04-04:
Thank you. How do I switch to CycloneDDS? In the screenshot above from the Create3 webserver, the box for “Enable Fast DDS discovery server?” is already unchecked.

BashByr, 2023-04-06:
@roni-kreinin Since changing this, I’ve been unable to view any topics on my PC. I have updated my `.bashrc` to include the downloaded cyclones_pc.xml file as mentioned in the manual:

```
alias humble='source /opt/ros/humble/setup.bash && export ROS_DOMAIN_ID=0 && export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp’

export CYCLONEDDS_URI=~/cyclonedds_pc.xml
```

But that didn’t change much. However, when I ssh into the RPI, I can see all topics published normally:

```
/parameter_events
/rosout
/tbot1/battery_state
/tbot1/cmd_vel
/tbot1/color/preview/image
/tbot1/diagnostics
/tbot1/diagnostics_agg
/tbot1/diagnostics_toplevel_state
/tbot1/dock_status
/tbot1/function_calls
/tbot1/hazard_detection
/tbot1/hmi/buttons
/tbot1/hmi/display
/tbot1/hmi/display/message
/tbot1/hmi/led
/tbot1/imu
/tbot1/interface_buttons
/tbot1/ip
/tbot1/joint_states
/tbot1/joy
/tbot1/joy/set_feedback
/tbot1/mouse
/tbot1/oakd/rgb/preview/camera_info
/tbot1/oakd/rgb/preview/image_raw
/tbot1/oakd/rgb/preview/image_raw/compressed
/tbot1/oakd/rgb/preview/image_raw/compressedDepth
/tbot1/oakd/rgb/preview/image_raw/theora
/tbot1/robot_description
/tbot1/scan
/tbot1/stereo/depth
/tbot1/tf
/tbot1/tf_static
/tbot1/wheel_status
```

Having said that, the TurtleBot’s green LEDs for ‘Comm’ and ‘Batt’ always turn off when I switch to CycloneDDS, and it no longer responds to its controller even though it connects to it.

Seeing these problems, I tried switching back to FastDDS by removing the extra lines in my `.bashrc` file and reconfiguring`turtlebot4-setup`. When I do that, the ‘Comm’ and ‘Batt’ LEDs come back on and I am able to control the robot with its controller again. On the RPI, I can also see that its publishing topics. However, I’m still unable to see any topics published through my PC. I’m not sure what switching to CycloneDDS might have changed that isn’t being reversed when I return to FastDDS.

BashByr, 2023-04-06:
@roni-kreinin I’ve switched to CycloneDDS and the LED/controller issue was fixed after reconnecting the Create3 to WiFi, thank you. However, I’m still unable to see any topics on my PC after restarting daemon. I’ve readded the two lines in my above reply back to my .bashrc file.

BashByr, 2023-04-07:
Just an update, the Create3 LEDs for BATT and COMM went back off and I am facing the same issues as my original post. This time, I’ve ensured that the Create3 is connected to WiFi, as I can see its IP address on the webserver and have been able to ping it successfully. I have tried rebooting and restarting the application through the webserver, and have restarted the Turtlebot4 service on the RPi. 

<details><summary>Create3 logs</summary>
<p>
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 syslog.info syslogd started: BusyBox v1.30.1
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.100652] spi-nand spi0.1: 128 MiB, block size: 128 KiB, page size: 2048, OOB size: 64
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.notice kernel: [    3.101010] 1 fixed-partitions partitions found on MTD device spi0.1
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.notice kernel: [    3.101018] Creating 1 MTD partitions on "spi0.1":
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.notice kernel: [    3.101037] 0x000000000000-0x000008000000 : "ubi"
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.191093] libphy: Fixed MDIO Bus: probed
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.191260] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.191266] ehci-platform: EHCI generic platform driver
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.201581] phy phy-10120000.usbphy.0: remote usb device wakeup disabled
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.201593] phy phy-10120000.usbphy.0: UTMI 16bit 30MHz
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.201617] ehci-platform 101c0000.ehci: EHCI Host Controller
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.201672] ehci-platform 101c0000.ehci: new USB bus registered, assigned bus number 1
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.202355] ehci-platform 101c0000.ehci: irq 26, io mem 0x101c0000
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223476] ehci-platform 101c0000.ehci: USB 2.0 started, EHCI 1.00
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223716] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223730] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223742] usb usb1: Product: EHCI Host Controller
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223753] usb usb1: Manufacturer: Linux 4.14.115-yocto-standard-custom ehci_hcd
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.223763] usb usb1: SerialNumber: 101c0000.ehci
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.224623] hub 1-0:1.0: USB hub found
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.224685] hub 1-0:1.0: 1 port detected
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.225419] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.225459] ohci-platform: OHCI generic platform driver
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.225738] ohci-platform 101c1000.ohci: Generic Platform OHCI controller
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.225772] ohci-platform 101c1000.ohci: new USB bus registered, assigned bus number 2
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.225866] ohci-platform 101c1000.ohci: irq 26, io mem 0x101c1000
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.297656] usb usb2: New USB device found, idVendor=1d6b, idProduct=0001
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.297677] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.297688] usb usb2: Product: Generic Platform OHCI controller
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.297698] usb usb2: Manufacturer: Linux 4.14.115-yocto-standard-custom ohci_hcd
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.297710] usb usb2: SerialNumber: 101c1000.ohci
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.298801] hub 2-0:1.0: USB hub found
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.298986] hub 2-0:1.0: 1 port detected
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.299840] usbcore: registered new interface driver cdc_acm
Apr  6 20:00:07 iRobot-8748D7279D6349E98FCA76C83EF27D99 kern.info kernel: [    3.299849] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[... cut ...]
             ├─1695 /opt/ros/humble/lib/tf2_ros/static_transform_publisher --x >
             ├─1729 /opt/ros/humble/lib/diagnostic_aggregator/aggregator_node ->
             └─1738 /usr/bin/python3 /opt/ros/humble/lib/turtlebot4_diagnostics>

Apr 07 03:59:44 ubuntu turtlebot4-start[1632]: [component_container-6] [INFO] [>
Apr 07 03:59:44 ubuntu turtlebot4-start[1632]: [component_container-6] [INFO] [>
Apr 07 03:59:44 ubuntu turtlebot4-start[1632]: [component_container-6] [INFO] [>
lines 1-25
</p>
</details>

roni-kreinin, 2023-04-10:
@BashByr Are you able to see all ros2 topics on the RPi still? Or are the Create 3 topics gone?

BashByr, 2023-04-19:
Thanks for all the help. After a few weeks of troubleshooting, this has been resolved. Not sure what restrictions my apartment WiFi had, but after adding a WiFi mesh router that connected to my home wifi through ethernet and connecting to that, everything now works great.

closed 2023-04-19 by BashByr

y2d2, 2023-07-31:
Hello, I tried the same as described here however I am not as lucky as BashByr.  Not sure if I can reopen the issue or have to start a new one? 

However, using cycloneDDS did not change the behavior wrt to using fastDDS using simple discovery. When using 2 robots with their unique namespace few minutes (Even seconds ) after boot up multiple topics disappear most importantly the "/ns"/odom topic stops with its respective transformation.
[image]

**Alternative tries**
I have tried out a lot of different things however without avail. 

- Setup simple discovery using fastDDS -> Not working as in this thread 
-  Setup 2 server Discoveries on both TB4s, not working due to changing the wired sub-mask net not correctly working see https://github.com/turtlebot/turtlebot4/issues/234 
- Tried to create a single discovery server on the controlling PC. Works fine for 1 TB, once the second one comes in same issues as described here. 

I am a bit a the end of the things I can think of on how to get it working. 

**Additional information**
Create3 firmware: H2.2 
ros version: humble 
installed: from repo 

Also, I see this issue is from 24th of April, yet the online documentation is just showing that connecting multiple tbs is as easy as just changing the namespace... https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html which it definitely is not...  at least changed the documentation to reflect the steps described here to change to simple discovery and use the cycloneDDS... which for me did not work.