turtlebot/turtlebot4 #133: Connection doesnt work

Cur1ousKea, 2023-04-04:
Hello, 

i got a Turtlebot 4 where someone already tried installing everything. But he couldnt finish and gave it to me. I have the issue now that i neither can connect the Turtlebot4 to the WiFi nor can i access it via LAN. One reason for that is that the LCD Display is pitchblack. When i try to access it via 192.168.185.3 (as in the documentation mentioned), it either says "No Route to Host" or "Connection refused".
I tried to reinstall Humble, but still nothing is showing up. 
Please Help and thank you for that in advance.

Best Regards 

Robert

roni-kreinin, 2023-04-05:
Which version of ROS2 are you using? Also what firmware is running on the Create 3?

From the logs it looks like the Create 3 is having some issues connecting to Wi-Fi. Has it been connected to a 2.4 GHz network?

Cur1ousKea, 2023-04-05:
Ive installed the latest available Version which is Humble 1.0.0. For Create 3 also the latest which is H1.1. It is connected to a 2.4Ghz Network. The same where the Raspi is connected to. But Raspi runs on 5Ghz. What makes me wonder is that the Battery status LED is blinking red. I dont think that this can be a WiFi issue ?

roni-kreinin, 2023-04-05:
These errors indicate Wi-Fi issues:
```
Apr 21 22:15:57 iRobot-0226BB1F284547CCB99759A95E828C63 daemon.notice network[1476]: [INFO] (src/gateway_pinger.c:113) wifi is disconnected
Apr 21 22:15:57 iRobot-0226BB1F284547CCB99759A95E828C63 daemon.warn network[1476]: [WARN] (src/gateway_pinger.c:132) no address count, 8
Apr 21 22:15:58 iRobot-0226BB1F284547CCB99759A95E828C63 user.notice ntpd: ntpd: send failed: Network unreachable
Apr 21 22:15:58 iRobot-0226BB1F284547CCB99759A95E828C63 user.notice ntpd: ntpd: send failed: Network unreachable
Apr 21 22:16:03 iRobot-0226BB1F284547CCB99759A95E828C63 user.notice create-platform: 1650579363.714025 [0]        tev: ddsi_udp_conn_write to udp/239.255.0.1:7400 failed with retcode -3
Apr 21 22:16:03 iRobot-0226BB1F284547CCB99759A95E828C63 user.notice create-platform: 1650579363.714131 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.14:7412 failed with retcode -3
```

There might also be a mismatch of RMW between the RPi and the Create 3 which will cause issues like the battery message publishing all 0's. In this case, the RPi would think that the battery is at a critical level even when its not, and force the robot to power off. You can check the actual battery level by going to the Create 3 Webserver, then going to the "About" section. There should be a "system battery" entry.

I would recommend going into the `turtlebot4-setup` tool and using the ROS Setup -> Bash setup menu to change the RMW_IMPLEMENTATION to `rmw_cyclonedds_cpp`. Then save the settings and return to the main menu to apply them.

Cur1ousKea, 2023-04-11:
I tried doing it but the results are the same. The result is the following: https://drive.google.com/file/d/19mmP9nJJaeKtymKM2Ec4BxGczpi50SpF/view?usp=share_link
 The funny thing is that when i started turtlebot after a few days, everything magically seemed fine without doing anything. I drove the robot around, it didnt shut off even when i let it stand for 5 minutes. Then i set it on the dock and after a few minutes it disconnected and shut off. Now i have the same issue again - it shuts of after 1 Min. Could there be a communication error with the controller? I noticed that every startup is different, in regards to the status leds. sometimes comm and batt will stay off the others are green. sometimes the battery blinks red until it gets removed from the dock(but shuts down no matter what). when i dont remove it and it continues blinking, it shut down after a while.:
https://drive.google.com/file/d/1ABRqQwv8Hu-pVPuRrgJKGNZhTTCIKdMN/view?usp=share_link 
But sometimes it also gets green on the dock...  After the shutdown the power button circles in red: https://drive.google.com/file/d/19yX48-hUk9dvhoJKx4fJCjeAqntrJfVv/view?usp=share_link
[messages (1).txt](https://github.com/turtlebot/turtlebot4/files/11200453/messages.1.txt)
[messages (2).txt](https://github.com/turtlebot/turtlebot4/files/11201243/messages.2.txt)
[messages (3).txt](https://github.com/turtlebot/turtlebot4/files/11201394/messages.3.txt)

Cur1ousKea, 2023-04-11:
I wanna add that the router is not connected to the internet. could that be a reason?

Cur1ousKea, 2023-05-17:
Sorry for the late response. I tried as you said, but theres a similar behaviour. https://drive.google.com/file/d/1Nbh53XwCHYJg55G3GMbw_YX2POwABRSC/view?usp=share_link
In the start it is booting, and sometimes the battery led is flashing red. after a while it goes to green.
I connected the controller, and it makes a small move and the controller isnt moving the robot at all. the only thing which works is the moving through the menu buttons. after that, the red status led is constant red. 
https://drive.google.com/file/d/1NPAt4ml_MlAA-gwlaS_9HAhWCPB0r0E0/view?usp=share_link
a short while after that this happens:
https://drive.google.com/file/d/1NMzAvdFD18klBQ99mY5smt7r1K6ZmKPT/view?usp=share_link

closed 2023-05-23 by Cur1ousKea

Cur1ousKea, 2023-08-22:
Hi there, it solved it for me once i completely reinstalled the turtlebot os. so erase the sd card and set everything up again. when u use sudo apt update && sudo apt upgrade let it run through, but once the window pops up where u have to press okay to restart the different services, hit cancel and manually reboot with sudo reboot