turtlebot/turtlebot4 #135: OAK-D Camera not working

shyamvanshikumar, 2023-04-05:
- OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed following the instructions that were present in the [TurtelBot4 - Basic Setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html)
 - Robot: TB4 standard

 I am trying to start the camera using the following command:
ros2 launch turtlebot4_bringup oakd.launch.py

It returns the following message and terminates:
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-04-05-16-34-12-787020-ubuntu-1992
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgb_stereo_node-1]: process started with pid [1994]
[rgb_stereo_node-1] [2023-04-05 16:34:16.963] [warning] Insufficient permissions to communicate with X_LINK_UNBOOTED device with name "1.1.2". Make sure udev rules are set
[rgb_stereo_node-1] [2023-04-05 16:34:17.415] [warning] Insufficient permissions to communicate with X_LINK_UNBOOTED device having name "1.1.2". Make sure udev rules are set
[rgb_stereo_node-1] terminate called after throwing an instance of 'std::runtime_error'
[rgb_stereo_node-1]   what():  No available devices
[ERROR] [rgb_stereo_node-1]: process has died [pid 1994, exit code -6, cmd '/opt/ros/galactic/lib/depthai_examples/rgb_stereo_node --ros-args --params-file /tmp/launch_params_yxbzj1h5 --params-file /tmp/launch_params_2nhhypkx --params-file /tmp/launch_params_zk32uh5i --params-file /tmp/launch_params_qvnkoo3k --params-file /tmp/launch_params__5plfs2j --params-file /tmp/launch_params_edkiucu7 --params-file /tmp/launch_params_59q6atll --params-file /tmp/launch_params_x064cmka --params-file /tmp/launch_params_85vmf3de --params-file /tmp/launch_params_16bjnd_m --params-file /tmp/launch_params_zuupmoka --params-file /tmp/launch_params_9q9n_7sg --params-file /tmp/launch_params_xx4_090i --params-file /tmp/launch_params__8hymvag'].

I am not sure what the problem is.
Thank you.

roni-kreinin, 2023-04-05:
@shyamvanshikumar Can you try running this on the RPi:
```
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```
Then launch oakd again.

shyamvanshikumar, 2023-04-05:
@roni-kreinin The camera is working now. Thank you.

closed 2023-04-06 by roni-kreinin

BradleyClymer, 2024-09-01:
@roni-kreinin Thank you for this answer -- helped me 17 months later when I was stuck, and I learned something for future development.