turtlebot/turtlebot4 #137: Connectivity issue

shyamvanshikumar, 2023-04-08:
Turtlebot4 rpi and create3
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 
User PC
-OS: Ubuntu 22.04 (running on a VM)
-ROS Distro: Humble

I have connect create3 to 2.4GHz band and rpi is connected to 5GHz band. I am using simple discovery protocols for getting all ros2 topics on the user pc. But none of the topic is visible on the user pc. All topics are visible on rpi though. I tried using both cyclonedds and fastdds but none worked.
 
List of topics on rpi:
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

List of topic on user pc:
/parameter_events
/rosout

Thank you

roni-kreinin, 2023-04-10:
You will need to update the TurtleBot4 to Humble to match your PC. Mixing versions of ROS 2 will cause communication issues.

See https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image and https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#updating-the-turtlebot-4

shyamvanshikumar, 2023-04-10:
I also tried using ROS2 Galactic (On Ubuntu 20.04) on my PC and I am still getting the same issue.

roni-kreinin, 2023-04-10:
Is your VM on the same subnet as the TurtleBot 4? You can check by calling `ip a` and checking the IP address.

closed 2023-09-21 by hilary-luo