turtlebot/turtlebot4 #141: Turtlebot4 lite Obstacle avoidance and performance issues

stalinADAQ, 2023-04-13:
1. I am currently using RPI4 with 4GB for turtlebot4 lite.
I could use navigation to move the robot from Point A to Point B, but now i would like to add obstacle avoidance while moving from A to B. How to do that ? is it already implemented and or need to be enabled ?
2. I understand that, there is way to view stereo depth camera from robot_view by enabling in the rviz tool, but it hanging entire process if we enable the stereo depth camera ? Is 4GB ram is not sufficient to monitor the camera ?
3. While running navigation using
"ros2 launch turtlebot4_navigation [localization.launch.py](http://localization.launch.py/) map:=office.yaml"
"ros2 launch turtlebot4_navigation [nav2.launch.py](http://nav2.launch.py/)"
it is always timed out waiting for odom server and unable fetch the odom topic.
How to resolve this issue as well ?

stalinADAQ, 2023-04-14:
1. But in my turtlebot4 lite setup, obstacle avoidance is not working ? I have followed the same documentation as [https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html]. Kindly advice 
3. The access point used here is mobile hotspot and operating only in 2.4GHz , is it having any performance problem ?

roni-kreinin, 2023-04-14:
1. Have you generated your own map? Which launch files are you using for navigating?
2. I would highly recommend using a 5GHz access point for the RPi4 especially if you are trying to stream video data. Even at the low resolution that the TurtleBot 4 uses, it can be quite taxing on network usage.

stalinADAQ, 2023-04-14:
@roni-kreinin yes, I am using my own map generated from SLAM.
1. 
   $ ros2 launch turtlebot4_navigation localization.launch.py map:=my_room.yaml
   $ ros2 launch turtlebot4_navigation nav2.launch.py
   $ ros2 launch turtlebot4_viz view_robot.launch.py
using above three command for starting the navigation. 
2. Another problem is even without camera, it is taking longer time activate navigation. after 5-10 minutes only it starting the navigation ?

roni-kreinin, 2023-04-14:
What are the issues with obstacle avoidance? Is the robot driving into obstacles that you can see on the map? As for the activation time, are there any errors being reported by either localization or nav2 in the terminal?

Also, are you running Galactic or Humble?

stalinADAQ, 2023-04-14:
Yes, it not detecting obstacles and trying drive into obstacles and failing to doso, if we manually clear the obstacle it going ahead.
We are using humble. I don't see any errors, but I will try to collect logs and share to you. 
Are the launch files are correct?

roni-kreinin, 2023-04-14:
The launch files are correct. Can you record your RViz while running navigation and share that here?

stalinADAQ, 2023-04-15:
@roni-kreinin please find the attached logs of the three terminal after running the navigation. I could not collect the lengthy log but it should give you insight of the problem.
[localization_launch.txt](https://github.com/turtlebot/turtlebot4/files/11238717/localization_launch.txt)
[nav2_launch.txt](https://github.com/turtlebot/turtlebot4/files/11238718/nav2_launch.txt)
[RVIZ.txt](https://github.com/turtlebot/turtlebot4/files/11238719/RVIZ.txt)

roni-kreinin, 2023-04-17:
From the RViz errors it looks like there is an issue with the time sync between the RPi and the Create 3. Try calling `turtlebot4-ntpd-sync` from the RPi4 or accessing the Create 3 webserver and restarting ntpd from the Beta Features tab.

stalinADAQ, 2023-04-20:
@roni-kreinin Thanks for pointing me in right direction, after turtlebot4-ntpd-sync, the obstacle avoidance is active in navigation and relatively the time for starting the navigation is less (still about 3-4 minutes of pose estimation). 
But one problem is, the moment obstacle is detected and not cleared for 3-5 sec, then the navigation in the RVIZ getting feed back as aborted. And hence it is not moving to new target. 

Also the initial pose delay is due to wi-fi speed ?

stalinADAQ, 2023-04-21:
@roni-kreinin please find the below link for screen recorded video, 

https://we.tl/t-sdS0ZHoqSt

I have uploaded in wetranfer due to size limitation. You can see the lag between pose and navigation and also obstacle abort problem at (9:22 minutes in video). Infact this time the response was much better :).

Please check and advise !

stalinADAQ, 2023-04-24:
@roni-kreinin waiting for your feedback, after reviewing above video.

stalinADAQ, 2023-04-25:
@roni-kreinin 
1. No, I am waiting until ros2 topic list shows all the topics published by create3 and then only running the navigation. 
But still it is taking 3-5 minutes to start the navigation.
2. During the course of booting the Create3 platform is showing RED in light ring for few seconds then giving chime sound. Is it normal ? because some times it is un responsive and not able to switch off on/off also ?
3. Yes, as i also mentioned in the video the obstacle behavior looks normal, but when there is obstacle present for little longer period, the navigation getting aborted. how to resolve this ?
Setup : ROS2 Humble 1.1 with Simple Discovery.
4. Also the turtlebot4-ntpd-sync need to be run every time we startup the ROBOT or only once enough ?
5. Still Am unable to use the Stereo camera at all, the moment it is enabled in RVIZ the entire process getting hanged.
6. Should i check Discovery server for better performance or simple discovery is good ?

hilary-luo, 2023-09-21:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-21 by hilary-luo

Vasanth28897, 2023-10-19:
@roni-kreinin  I just want to know, by how the obstacle avoidance works in the turtlebot4, when i run these commands...
`ros2 launch turtlebot4_navigation nav2.launch.py `
`ros2 launch turtlebot4_navigation localisation.launch.py map:=my_map.yaml`
`ros2 launch turtlebot4_viz view_robot.launch.py`

with the help of RPI-Lidar or OAK-D camera or both?

hilary-luo, 2023-10-25:
@Vasanth28897 The short answer is just the lidar. The long answer is that those commands that you are referring to are launching Nav2. There are a lot of things that you can do with [Nav2](https://navigation.ros.org/). All of the default Nav2 settings are contained in the [config yaml files](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config). You can modify this behaviour by making your own yaml files and passing them in when you launch nav2 and/or localization.

Vasanth28897, 2023-10-28:
thank you @hilary-luo.  I am facing the same obstacle avoidance problem in the turtlebot4 i have when i run the nav2.launch.py like @stalinADAQ and @schenkch faced. I tried the time-sync command...it didn't work.

hilary-luo, 2023-11-10:
@Vasanth28897 I am going to need you to expand on what you mean when you are referring to the "same obstacle avoidance problem". Preferably can you do this in a fresh ticket? Time has changed since then, the software has been updated in various ways so that symptom may have a different reason now.

Vasanth28897, 2023-11-16:
Alright, as @hilary-luo mentioned in  issue. The lidar cannot hits the costmap around the mapped dynamic obstacle(not in the map during generation)as you mentioned in [this ](https://github.com/turtlebot/turtlebot4/issues/190#issuecomment-1698050402) issue. (Note: sometimes it hit the static obstacle also).

hilary-luo, 2023-11-16:
@Vasanth28897 If I am understanding correctly, you are seeing that the robot is not avoiding dynamic obstacles. This issue is fixed in 1.0.4. You can use it now if you build from source, or it will be released in the next update. The issue and the temporary fix were tracked here: https://github.com/turtlebot/turtlebot4/issues/247

If it is hitting static obstacles then either a) it had issues seeing them (not at lidar height or are too skinny, too reflective or see through etc.) and you may need to block off those areas or if it is not the case, b) you can tune your nav2.yaml parameters to better work for your environment. Different environments and different desired operations will require different parameters.

Vasanth28897, 2023-11-17:
alright, thank you @hilary-luo for the clarification.