turtlebot/turtlebot4 #142: Automatic shutdown

ddwy, 2023-04-16:
Hello, I am using the image of the turbobot4 lite humble version. After starting create3 and Raspberry Pie, after a few minutes, it will automatically shut down.Why did this happen?

roni-kreinin, 2023-04-17:
Is the robot docked when this happens? Also what is the battery level at? 

The `turtlebot4_node` starts a 60s power off timer when the Create 3 reaches 12% battery. If it is not placed back on a dock in those 60s then it will shut itself off to protect the battery.

ddwy, 2023-04-18:
> @ddwy I have the same issue as well. I have observed my turtlebots rebooting itself after few minutes.

The problem has been resolved. After using the turbobot4 hub version, create3 also needs to update the firmware

closed 2023-04-20 by roni-kreinin