turtlebot/turtlebot4 #143: Controller failed to pair

languyen14507, 2023-04-16:
Hello,

I have just moved my Turtlebot4 to a different network. I was able to set up the wifi communication with the RPi and Create@3 but could not pair the controller. These are the messages:

[bluetooth]# exit
ubuntu@ubuntu:~$ sudo bluetoothctl --agent=NoInputNoOutput
Agent registered
[CHG] Controller E4:5F:01:BD:05:2B Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller E4:5F:01:BD:05:2B Discovering: yes
[CHG] Device A0:5A:5C:EC:51:07 RSSI: -56
[CHG] Device A0:5A:5C:EC:51:07 Modalias: usb:v054Cp09CCd0100
[bluetooth]# trust A0:5A:5C:EC:51:07
Changing A0:5A:5C:EC:51:07 trust succeeded
[bluetooth]# pair A0:5A:5C:EC:51:07
Attempting to pair with A0:5A:5C:EC:51:07
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# pair A0:5A:5C:EC:51:07
Attempting to pair with A0:5A:5C:EC:51:07
Failed to pair: org.bluez.Error.AlreadyExists
[bluetooth]# connect A0:5A:5C:EC:51:07
Attempting to connect to A0:5A:5C:EC:51:07
Failed to connect: org.bluez.Error.Failed

Please help. Thank you.

schenkch, 2023-04-17:
I had a similar issue. I got it connected and paired again using this https://github.com/turtlebot/turtlebot4/issues/14
but  I cannot get it to auto-connect any more by just pushing the home button.  Every time it disconnects again, I have to remove it, and then reconnect it again using sudo bluetoothctl. Is there a way that I can get it to connect by pushing the home button again?

roni-kreinin, 2023-04-17:
Can you try pushing the reset button on the back of the controller, then trying again?

languyen14507, 2023-04-18:
I pushed the reset button, followed the suggestion by schenkch to remove the controller, then was able to re-pair and reconnect the controller. But the connection is lost quickly before I have enough time to try to undock and move the turtlebot4. I again tried this cycle (remove, re-pair and reconnect the controller) many times and every time the controller lost the connection quickly after a successful connection. Please see the screen history below

ubuntu@ubuntu:~$ sudo bluetoothctl --agent=NoInputNoOutput
Agent registered
[CHG] Controller E4:5F:01:BD:05:2B Pairable: yes
[bluetooth]# devices
Device A0:5A:5C:EC:51:07 Wireless Controller
Device 68:04:0A:AC:06:93 68-04-0A-AC-06-93
Device 39:12:BC:22:B3:74 39-12-BC-22-B3-74
[bluetooth]# remove A0:5A:5C:EC:51:07
[DEL] Device A0:5A:5C:EC:51:07 Wireless Controller
Device has been removed
[bluetooth]# untrust A0:5A:5C:EC:51:07
Device A0:5A:5C:EC:51:07 not available
[bluetooth]# exit
ubuntu@ubuntu:~$ 
ubuntu@ubuntu:~$ 
ubuntu@ubuntu:~$ sudo bluetoothctl --agent=NoInputNoOutput
Agent registered
[CHG] Controller E4:5F:01:BD:05:2B Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller E4:5F:01:BD:05:2B Discovering: yes
[CHG] Device 39:12:BC:22:B3:74 RSSI: -70
[CHG] Device 68:04:0A:AC:06:93 RSSI: -73
[CHG] Device 68:04:0A:AC:06:93 TxPower: 6
[NEW] Device A0:5A:5C:EC:51:07 Wireless Controller
[bluetooth]# trust A0:5A:5C:EC:51:07
[CHG] Device A0:5A:5C:EC:51:07 Trusted: yes
Changing A0:5A:5C:EC:51:07 trust succeeded
[bluetooth]# pair A0:5A:5C:EC:51:07
Attempting to pair with A0:5A:5C:EC:51:07
[CHG] Device A0:5A:5C:EC:51:07 Connected: yes
[CHG] Device A0:5A:5C:EC:51:07 Modalias: usb:v054Cp05C4d0100
[CHG] Device A0:5A:5C:EC:51:07 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device A0:5A:5C:EC:51:07 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device A0:5A:5C:EC:51:07 ServicesResolved: yes
[CHG] Device A0:5A:5C:EC:51:07 Paired: yes
Pairing successful
[CHG] Device A0:5A:5C:EC:51:07 ServicesResolved: no
[CHG] Device A0:5A:5C:EC:51:07 Connected: no
[bluetooth]# connect A0:5A:5C:EC:51:07
Attempting to connect to A0:5A:5C:EC:51:07
[CHG] Device A0:5A:5C:EC:51:07 Connected: yes
Connection successful
[CHG] Device A0:5A:5C:EC:51:07 ServicesResolved: yes
[NEW] Device 45:1A:06:54:25:15 45-1A-06-54-25-15
[NEW] Device 23:34:71:BC:85:6C 23-34-71-BC-85-6C
[CHG] Device A0:5A:5C:EC:51:07 ServicesResolved: no
[CHG] Device A0:5A:5C:EC:51:07 Connected: no
[bluetooth]# connect A0:5A:5C:EC:51:07
Attempting to connect to A0:5A:5C:EC:51:07
Failed to connect: org.bluez.Error.Failed
[DEL] Device 39:12:BC:22:B3:74 39-12-BC-22-B3-74
[DEL] Device 68:04:0A:AC:06:93 68-04-0A-AC-06-93
[DEL] Device 45:1A:06:54:25:15 45-1A-06-54-25-15
[DEL] Device 23:34:71:BC:85:6C 23-34-71-BC-85-6C
[NEW] Device 23:34:71:BC:85:6C 23-34-71-BC-85-6C
[NEW] Device 45:1A:06:54:25:15 45-1A-06-54-25-15
[bluetooth]# exit

schenkch, 2023-04-18:
Yes, I tried pushing the reset button as well and can get it connected like @languyen14507 but after some time not using the controller or after reboot I have to do the same thing again and cannot get it to auto-connect to the same hardware ID with the home button any more.

languyen14507, 2023-04-19:
Is there a way to dock/undock the Turtlebot4 using the computer keyboard with "ros2 run teleop_twist_keyboard teleop_twist_keyboard"? I use the Turtlebot4 for demos a lot so this would be a great backup whenever the controller acts up. Thank you so much.

tschinfra, 2023-04-19:
Hey there i have the same issue like @languyen14507 and @schenkch. But with the difference that the controller is disconnected immediately after it was connected. I followed the steps in the [documentation](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html) and also tried resetting the controller als well as #14.
Are there any additional steps one could try?

Initially i wanted to update the robot from Galactic to Humble and noticed that the controller wasn't connected anymore (after successfully updating). Could this have something to do with it?

roni-kreinin, 2023-04-19:
> Is there a way to dock/undock the Turtlebot4 using the computer keyboard with "ros2 run teleop_twist_keyboard teleop_twist_keyboard"? I use the Turtlebot4 for demos a lot so this would be a great backup whenever the controller acts up. Thank you so much.

@languyen14507 The dock/undock feature is a ros2 action. You can perform it from CLI by calling `ros2 action send_goal /dock irobot_create_msgs/action/Dock {}` or `ros2 action send_goal /undock irobot_create_msgs/action/Undock {}` . Additionally the turtlebot4 node maps a short press of Create 3 button 1 to the dock action, and a short press of button 2 to the undock action.

Regarding the controller issues, I will try to recreate it and find a fix. I would also recommend making sure that the controllers are fully charged just in case.

languyen14507, 2023-04-23:
Hello Roni, I have run into another problem. In order to communicate with the Turtlebot 4 from my PC via a wifi router, does the router have to also be accessible to the internet? I changed my network environment  from one where the router has the internet to one where the router does not have the internet (due to the university security requirements). In the network where the router has the internet access, everything works fine, I can connect the controller, undock the turtlebot and drive it around. In the network where the router does not have the internet access, I cannot undock the turtlebot with the controller or with the "ros2 action send_goal undock ..." command.  I can ssh to the RPi and connect to the Create@3 but nothing else would work. Please help, I am scheduled to have a demo very soon. Thank you so much.

languyen14507, 2023-04-25:
@roni-kreinin I haven't been able to check the synchronising time between the RPi4 and Create@3 yet because I tried to reboot the Turtlebot 4 but could not. I removed the turtlebot from dock, pressed and held the power button, the light flashed 3 times but stayed on and the turtlebot did not shutdown. I repeated this process many times and could not power off the turtlebot. Sometimes, the light would go off a few seconds after flashing 3 times then it comes right back without any sound. Sometimes there is a sound but the turtlebot would not power off. Is there another way to force the Turtlebot 4 to power off without having to wait until the battery is completely drained?

roni-kreinin, 2023-04-26:
@languyen14507 If you press and hold the power button for 20-30s it will perform a full reboot on the Create 3 which should fix that issue.

languyen14507, 2023-04-26:
The time on  Creat@3 is behind by about 5 days.

On RPi
languyen:~$ date
Wed 26 Apr 2023 03:11:58 PM CDT

On Create@3
Apr 21 13:28:03 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: 1650547683.354310 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.4:38455 failed with retcode -1

How do we fix that?

languyen14507, 2023-04-27:
@roni-kreinin   I just noticed that I copied from the wrong terminal for RPi4, so here are the correct times

On RPi4:
ubuntu@ubuntu:~$ date
Thu 21 Apr 2022 07:03:46 PM UTC

On Create@3:
Apr 27 01:47:33 iRobot-02D154EDE40449C5845F2263ADB1EFA0 user.notice create-platform: 1682560053.954181 [0]        tev: ddsi_udp_conn_write to udp/192.168.1.4:49991 failed with retcode -1

So it looks like they don't have a synchronized time. Would this be the cause for me not able to undock the Turtlebot 4 with the controller and what do I need to do to fix it? Thanks.

roni-kreinin, 2023-05-01:
It is possible. You will need to either use your PC as a time sync source for the RPi, or manually set the time on the RPi.

To manually set time, call `sudo timedatectl set-ntp false` then `sudo timedatectl set-time '2015-11-20 16:14:50'` with the correct date/time.

languyen14507, 2023-05-01:
@roni-kreinin  I did reset the time on the RPi  to within a few seconds of the Create@3's as shown in the screen shot below. I also got the controller connected but still could not undock or move the Turtlebot 4. There appear to be some error messages in orange and red color on the Create@3 login screen. Could you please tell me what is still wrong? Thanks.
[image]

roni-kreinin, 2023-05-03:
Can you try configuring your turtlebot4 to use the discovery server as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)? It seems that the Create 3 is having issues connecting to your new network. With the discovery server, only the RPi4 needs to connect to Wi-Fi.

schenkch, 2023-05-11:
Are there any news on the issue that the controller has to be connected manually any more and is not connected automatically any more by pushing the home button? 
Also, I noticed that when pushing the home button to send it back to its base it moves a little bit but does not go back to home unless it is very close to it. Are these two issues related or any idea what could be the fix for this?

tschinfra, 2023-05-24:
I have a little update to the controller issue @schenkch @roni-kreinin.

Several times now i tried to connect the original controller to the TB4 but without success. Then I tried connecting a Playstation 4 controller and it worked right away! I was following the standard procedure as mentioned [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#turtlebot-4-controller-setup).

So there could be an issue regarding the original controller the TB4 comes with...

languyen14507, 2023-06-23:
@roni-kreinin The process of reconfiguring the Turtlebot 4 to use the discovery server is somewhat complicated so I would like to try using our university private WiFi network to run Turtlebot 4 first since it does have internet. After logging into RPi via AP mode, if I "sudo wifi.sh -s 'PRIVATE_WIFI' -p 'password' -r US && sudo reboot", will an IP address be assigned to the RPi and be displayed on the LED? If for some reason, this does not happen and/or I no longer able to ssh into the RPi, how can I get the RPi back into AP mode? Thanks.

closed 2023-09-21 by hilary-luo