turtlebot/turtlebot4 #145: Multiple robot support with discovery server

roni-kreinin, 2023-04-17:
**Describe the the feature you would like**
With H.1.2 we can now change the `usb0` subnet (https://github.com/iRobotEducation/create3_docs/issues/335). This should allow for one PC to communicate with multiple robots running discovery server.

Changes needed:
- Add menu option to `turtlebot4_setup` to change the `usb0` subnet on the RPi4 and Create 3.
- Update [configure_discovery.sh](https://github.com/turtlebot/turtlebot4_setup/blob/humble/turtlebot4_discovery/configure_discovery.sh) to support multiple discovery servers and varying subnets.

Notes:
- Reserve the `192.168.185` subnet for `eth0`

dmorris0, 2023-10-20:
This functionality would be very useful!

kemsulli1, 2024-01-11:
I have been working on this.  I can successfully change the IP address on the Create (and changed the Discovery Server IP to match), and and see the ROS topics on the host PC (I replaced all references of 192.168.186.3 with the new IP address).  I can succesfully add a route to a second robot.  However, I can only see data and control one robot.  I have tried changing the ROS_DOMAIN_ID and ROS_DISCOVERY_SERVER on the host PC; didn't work.  I have not yet tried namespaces.   Anyone have any other thoughts?

hilary-luo, 2024-01-15:
@kemsulli1 You do need to use namespaces when setting up multi-robot unless you put all of the robots on their own ROS_DOMAIN_IDs. See this page: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/multiple_robots.html#namespacing

closed 2024-06-27 by hilary-luo

m-usama-z, 2024-06-29:
Hi. In order to utilize the multi-robot discovery server support, do we simply repeat the existing instructions for discovery server setup on both the robots and the single pc being used to access the robots, for each robot, or are there any new instructions to follow for this? We have been unable to move more than 3 robots concurrently in simple discovery and will highly appreciate any help. Thank you.

hilary-luo, 2024-06-30:
Either you need to build this from source and use the instructions in the corresponding user manual PR or wait until both the apt packages are live and the user manual PR is merged and released.

m-usama-z, 2024-06-30:
I see the new user manual, thank you. As for the apt packages, I assume you are talking about the turtlebot4-setup and turtlebot4-bringup apt packages?

m-usama-z, 2024-07-01:
I was unable to properly build turtlebot4-setup 😅 I also tried copying the newly modified files manually to the corresponding directories in the turtlebot's file system and then following the user manual's instructions, but it did not work. Can you please provide an estimate of when the apt packages might go live? Thanks again.

hilary-luo, 2024-07-02:
The packages are going through the release process right now so it shouldn't be long and they will be available on the test servers. Then it will just be a matter of the ROS 2 Humble package sync before they are available on the main ROS servers. Unfortunately both of those processes involve interactions between multiple people from different organizations so I cannot tell you when they will be available. You can keep an eye open here: https://discourse.ros.org/tag/humble for the next sync ("new packages") announcement. Generally syncs happen at least once a month.