turtlebot/turtlebot4 #146: Compatibility and availability galactic and humble

schenkch, 2023-04-19:
I started using the Turtlebot4 with Ubuntu 22.04 Humble on my remote PC and on the RPi there was Galactic preinstalled. Are there any compatibility issues with that? I am debating whether it would be better to upgrade the RPi or use Ubuntu 20.04 Galactic on the remote PC. I saw that for humble the simulation package is not yet available. Is this the only package or are there other features that are not yet available? Are there any hardware recommendations or requirements for the remote PC especially when using the camera?

roni-kreinin, 2023-04-19:
You will likely run into issues with using Humble on your PC and Galactic on the robot. I would recommend upgrading the robot to Humble. The simulation is the only package not released yet, we are waiting on a PR to be merged before releasing that.

As for hardware, most PC's should be able to handle things like RViz, SLAM and Nav2 just fine. I would recommend having a dedicated GPU if you are going to be using the simulator.

schenkch, 2023-04-19:
Great, thanks. And is there a way I can do a backup before or from experience upgrading the robot and RPi from Galactic to Humble does not give any issues?

tschinfra, 2023-04-20:
@schenkch Yes, you can look [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi) and search for the chapter 'Install latest Raspberry Pi image'. There is the command `sudo dd if=/dev/<SD_NAME> of=<IMAGE_PATH> bs=1M` to back up the image.

schenkch, 2023-05-11:
Can you tell me what the easiest way is to get to the SD card with a Turtlebot4 Standard ? Do I have to disassemble the top or can I get to it somehow through the tray on the bottom? Are there any instructions any where?

roni-kreinin, 2023-05-15:
@schenkch You will have to remove the top plate and standoffs, then lift off the PCBA to access the SD card. Instructions are available [here](https://turtlebot.github.io/turtlebot4-user-manual/mechanical/turtlebot4.html#removing-the-pcba).

22222140, 2023-05-16:
Hello, I'm also working on my Turetlebot4 with Ubuntu 22.04 Humble on my remote PC.
I'm using another SD card with Ubuntu 22.04 (for RPi4) and installed Humble as well. But cannot connect it to Create3 Webserver.
Can I get a solution for it? (I could setup SSH and Wifi without Create3 webserver.)

Actually I tried to upgrade to Humble with provided SD card, but it was not possible because it's focal, not jammy.

schenkch, 2023-05-16:
Great, thank you @roni-kreinin!

roni-kreinin, 2023-06-01:
@22222140 Can you try installing the turtlebot4 humble image from [here](http://download.ros.org/downloads/turtlebot4/)? Installation instructions are [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image).

closed 2023-09-21 by hilary-luo