turtlebot/turtlebot4 #147: Create3 fw H.1.2. uses 100% CPU

JasmineRosselliSUPSI, 2023-04-20:
- OS: Ubuntu 22.04.1
 - ROS Distro: Humble
 - Installed
 - Package version: H.1.2. (iRobot Create3)

 **Expected behaviour**
It is expected that the robot normally boot (white light until 'happy sound') then ready to perform without excessive CPU usage.

 **Actual behaviour**
At boot, after a couple seconds of white light, the robot gives a solid red light for a few tens of seconds, then connects to WiFi (takes more time that usual) and finally reboot as usual.

After doing some other task on turtlebot4, e.g. launching navigation, the log of the Create3 indicates a CPU usage of 100%.
-> Turtlebot4 slower to react after updating to Humble.

There is the attached log of Create3:
[log.txt](https://github.com/turtlebot/turtlebot4/files/11287486/log.txt)

**Could this affect normal Turtlebot4 efficiency?**

roni-kreinin, 2023-04-20:
This is a known issue with FastRTPS on the Create 3 in Humble. iRobot is aware and working on it.

In the meantime I would recommend using either the Discovery server or CycloneDDS.

JasmineRosselliSUPSI, 2023-04-21:
Thank for your reply.
Actually we where already use CycloneDDS, in Simple Discovery mode. The attached log was in fact generated using this configuration.

Is then possible that this issue is not only related to the DDS.

sashag1991, 2023-07-04:
Hi,

Do you have some estimation when the fix for CPU utilization using humble (with iRobot Create3 platform) should arrive ?
I need to my work, and no I am stuck because the robot don't avoid obstacles due to CPU utilization (https://github.com/turtlebot/turtlebot4/issues/190)

Thanks.

JasmineRosselliSUPSI, 2023-07-05:
Hi,

Actually my colleagues and I returned to Galactc, because we could not afford to waste time trying to fix the issue.
We noticed that even if we went back to Galactic the CPU usage still seems to reach 100% and Turtlebot4 seems to react more slowly than before the update to Humble. Perhaps we forgot to reset some variables as its original Galactic value, but we are not sure. (While in Humble, we also had the following issue: Humble AMCL Particle Swarm doesn't work #148 )

Has anyone managed to solve the issue?

mdxtinkernick, 2023-07-08:
there is a new firmware release for the robot base - should we wait for a pi image update, or just run apt update on the pi ?

sashag1991, 2023-07-09:
Is there any workaround to the CPU utilization issue ?
Change to galactic maybe ?

mdxtinkernick, 2023-07-09:
I was hoping there may be, but on looking more closely it appears to have no major bug fixes, but does add config options for ntp and xml configuration configuration of middleware, which may provide ways to fix things, but not an explicit fix. It may be an issue that needs some work from the dds side as well as from icreate

marredir, 2023-08-29:
Do we know if this bug is related to the fw on the create3 base or on ubuntu/raspberry side? If I look [here](https://iroboteducation.github.io/create3_docs/releases/overview/) it says: "As of March 2023, use of Galactic is recommended, as it is more performant than Humble on Create 3." By saying "_more performant_"  do you mean that you don't get the aforementioned bug or does it mean that anyway the fw can be used?

Is there active work on fixing the bug or does iRobot plan to focus exclusively on Galactic from now on?

marredir, 2023-08-31:
Another question: do we know if there is a difference in having the Create3 directly connected to the Wifi or handling the network communication via USB? In other terms, can it be either (Wifi or USB) configuration the problem or do we have the same issue no matter what? Thanks

hilary-luo, 2023-09-14:
@marredir There are many pieces involved in this issue, it includes both the create3 (including firmware) and the raspberry pi. I can't speak for iRobot and exactly what they mean in their documentation. What I can say is that there is active effort going on to better define the 100% CPU issue and to address it completely. If anyone does want to participate in the testing part of this process feel free to let me know. (Involves giving detailed information about your setup and what you are trying to achieve and then trying changes and giving detailed descriptions of the changes)

There are definitely differences between CPU usage with discovery server vs simple discovery. I would encourage you to try discovery server and share whether it helps for you. That would be helpful information for me to have. I think it should help.

marredir, 2023-09-15:
Hi @hilary-luo ,

thanks for the reply. I can confirm that using the discovery server improves things in Galactic (i.e. it is less unstable but still not very reliable). I am interested in getting an understanding if we can switch permanently to Humble so I am definitely willing to help (and from this point of view may be you have more "profiling" info than me). Please contact me via email and we can exchange more information/instructions there.

Thanks

marredir, 2023-09-27:
@hilary-luo should I contact you by email with a description of my setup and then you can start to discuss how to go further?
Thanks

hilary-luo, 2023-10-02:
Yes @marredir, go ahead. That would be the fastest start to this. I have been very busy so I will respond to it as soon as I am able.

marredir, 2023-11-06:
Please refer to [this](https://github.com/turtlebot/turtlebot4/issues/266) for further discussion

hilary-luo, 2023-11-10:
Closing this issue since we are moving discussion to the other thread as mentioned and to avoid duplication.

closed 2023-11-10 by hilary-luo