turtlebot/turtlebot4 #148: Humble AMCL Particle Swarm doesn't work

JasmineRosselliSUPSI, 2023-04-20:
- OS: Ubuntu 22.04.1
 - ROS Distro: Humble
 - Installed
 - (Package version: H.1.2. Create3)
 - CycloneDDS, Simple Discovery mode

 **Expected behaviour**
 Normally in RViz it is possible to see the estimated locations given by Amcl Particle Swarm.

 **Actual behaviour**
After starting RViz the Amcl Particle Swarm doesn't work.
LaserScan works only sometimes, otherwise the error is: 
> Could not transform from [rplidar_link] to [map]

**To Reproduce**
Runned code from Turtlebot4 (ssh connection)
`ros2 launch turtlebot4_navigation localization.launch.py map:=my_map.yaml`

Other terminal:
`ros2 launch turtlebot4_navigation nav2.launch.py`

Launched RViz from an Humble PC:
`ros2 launch turtlebot4_viz view_robot.launch.py`

 
**Please help!**

[image]

JasmineRosselliSUPSI, 2023-04-21:
Here is the RViz terminal messages:
[image]

For additional informations, we attach the message generated by localization and navigation terminals:
Localization: (`ros2 launch turtlebot4_navigation localization.launch.py map:=my_map.yaml`)
[image]

Navigation: (`ros2 launch turtlebot4_navigation nav2.launch.py`)
[image]

ddwy, 2023-04-24:
I have also encountered this problem, have you solved it?

closed 2023-09-21 by hilary-luo