turtlebot/turtlebot4 #149: PRLIDAR-A1 only returns infinity for ranges

tilly111, 2023-04-21:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Galactic

 **Expected behaviour**
The LIDAR scan topic of the Turtlebot 4 should return distances and intensities other than .inf and 0.0 respectively. (Tested with another Turtlebot 4 in same environment - there it worked as expected).

 **Actual behaviour**
The LIDAR scan message looks like:
```
---
header:
  stamp:
    sec: 1682065034
    nanosec: 651834082
  frame_id: rplidar_link
angle_min: -3.1241393089294434
angle_max: 3.1415927410125732
angle_increment: 0.006065567955374718
time_increment: 0.0001247737236553803
scan_time: 0.128891259431839
range_min: 0.15000000596046448
range_max: 12.0
ranges:
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- .inf
- '...'
intensities:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- '...'
---
```

**To Reproduce**
Start the Turtlebot 4. Type `ros2 topic echo /scan`. Could it be a hardware problem, i.e., the LIDAR is broke. 

**Other notes**
Controlling the LIDAR, e.g., turning the motor off and on works fine (while turned off no messages are published).

tilly111, 2023-04-26:
It does not show anything in rViz and also no change in values when I move my hand around it.

roni-kreinin, 2023-04-26:
Can you stop the `turtlebot4` service with `sudo systemctl stop turtlebot4`, then manually launch the RPLIDAR with `ros2 launch turtlebot4_bringup rplidar.launch.py`? Do any errors show up when launching it?

closed 2023-09-21 by hilary-luo