turtlebot/turtlebot4 #150: Turtlebot 4 scan data not available

binni08, 2023-04-24:
Hello.

I'm using Turtlebot 4, but when I turn on the robot, the scan value for the rplidar does not come out.

The topic is issued, but the inf value is showing up in the data, and it doesn't seem to have an error separately.

Even if we disconnect and re-insert the usb cable for rplidar, we still can't check scan data in rviz.

Please check.
[image]
[image]

roni-kreinin, 2023-04-24:
It looks like you are getting RPLIDAR data. `inf` just means that the range for those points exceeds the max range of the RPLIDAR. Since you are not publishing the robot model, RViz does not know how to transform the laser scan data to `base_link`. Try changing your Fixed Frame to `rplidar_link`.

binni08, 2023-04-26:
Hello,

I floated the robot model and checked the laser scan data, and it seems that only the data comes out in a certain part, not 360 degrees.

Even if you change the position of the box located in the front, the scan data is not coming out.

Please refer to the attached video.

Thank you.

[simplescreenrecorder-2023-04-26_10.46.54.zip](https://github.com/turtlebot/turtlebot4/files/11328184/simplescreenrecorder-2023-04-26_10.46.54.zip)

roni-kreinin, 2023-04-26:
Which ROS2 version are you running? Also have you made any changes to the RPLIDAR launch file?

binni08, 2023-04-27:
Yes, I am currently using the galactic version and there are no files modified.

roni-kreinin, 2023-04-28:
Can you try running the RPLIDAR from your PC and check if the issue persists?

You can install the package with `sudo apt install ros-galactic-rplidar-ros`, then call `ros2 launch rplidar_ros rplidar.launch.py` with the RPLIDAR plugged in.

binni08, 2023-05-02:
The problem is that the rider scan data does not appear in the but is less than the actual obstacle.

If you look at the video that I sent you before, you can see that it's actually not scanned even though the obstacle is gone.

The rplidar_ros package is already installed, and the result is the same if you unplug and reseat the USB and run the rider alone.

binni08, 2023-05-03:
When I checked with a normal robot in the same space, the scan data came out as shown in picture 1 attached, and the scan for the problem robot is shown in picture 2.

[image]
[image]

roni-kreinin, 2023-05-03:
This might be a faulty RPLIDAR. Please contact support@clearpathrobotics.com so that we can replace it.

closed 2023-09-21 by hilary-luo