turtlebot/turtlebot4 #151: LCD screen off and RaspberryPi does not connect to network

morpheus1820, 2023-04-26:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed from factory

 **Expected behaviour**
RaspberryPi4 boots on and connects to network.

 **Actual behaviour**
Robot suddenly stopped working.
The LCD **screen on the turtlebot does not turn on**; RaspberryPi4 computer **does not seem to boot** (when connecting via microHDMI, all I can see is a rainbow screen on both HDMI outputs). **Neither WIFI or ethernet devices are accessible** (the default eth0 ip, which is specified correctly in the netplan config, is not reachable. So is the wifi ip, which is also configured correctly). 
The Create3 base, instead, seems to correctly connect  to the same WIFI network as before.
I wonder what would be the issue or how to debug it.

roni-kreinin, 2023-04-26:
That does sound like either the RPi4 has died or the SD card got corrupted somehow. Did this occur while the robot was running or when you tried powering it on?

You can try to re-image the SD card by following these instructions: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

morpheus1820, 2023-04-26:
Thank you. It was indeed the SD card that got corrupted, re-flashing fixed the issue. The trouble is that the SD card got corrupted after a normal shutdown.

closed 2023-04-26 by morpheus1820