turtlebot/turtlebot4 #153: Can not drive the turtlebot4

tianmuw, 2023-04-27:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble

 **Expected behaviour**
I tried to use the keyboard to move the turtlebot, running the teleop_twist_keyboard command both on the PC and the turtlebot4, but can not work, there is no response when I pressed keys.
Also, I tested the ros2 action send_goal command for the undock action and the drive_distance action, it showed that "Waiting for an action server to become available". What is this action server, shouldn't it automatically be there?

tianmuw, 2023-05-01:
The ros2 topic list on the Turtlebot4:
/parameter_events
/rosout
/yosemite/battery_state
/yosemite/cmd_audio
/yosemite/cmd_lightring
/yosemite/cmd_vel
/yosemite/color/preview/image
/yosemite/diagnostics
/yosemite/diagnostics_agg
/yosemite/diagnostics_toplevel_state
/yosemite/dock_status
/yosemite/function_calls
/yosemite/hazard_detection
/yosemite/hmi/buttons
/yosemite/hmi/display
/yosemite/hmi/display/message
/yosemite/hmi/led
/yosemite/imu
/yosemite/interface_buttons
/yosemite/ip
/yosemite/ir_intensity
/yosemite/ir_opcode
/yosemite/joint_states
/yosemite/joy
/yosemite/joy/set_feedback
/yosemite/kidnap_status
/yosemite/mobility_monitor/transition_event
/yosemite/mouse
/yosemite/odom
/yosemite/robot_description
/yosemite/robot_state/transition_event
/yosemite/scan
/yosemite/slip_status
/yosemite/static_transform/transition_event
/yosemite/stereo/depth
/yosemite/stop_status
/yosemite/tf
/yosemite/tf_static
/yosemite/wheel_status
/yosemite/wheel_ticks
/yosemite/wheel_vels

The ros2 topic list on the PC:
/parameter_events
/rosout
/yosemite/battery_state
/yosemite/cmd_audio
/yosemite/cmd_lightring
/yosemite/cmd_vel
/yosemite/color/preview/image
/yosemite/diagnostics
/yosemite/diagnostics_agg
/yosemite/diagnostics_toplevel_state
/yosemite/dock_status
/yosemite/function_calls
/yosemite/hazard_detection
/yosemite/hmi/buttons
/yosemite/hmi/display
/yosemite/hmi/display/message
/yosemite/hmi/led
/yosemite/imu
/yosemite/interface_buttons
/yosemite/ip
/yosemite/ir_intensity
/yosemite/ir_opcode
/yosemite/joint_states
/yosemite/joy
/yosemite/joy/set_feedback
/yosemite/kidnap_status
/yosemite/mobility_monitor/transition_event
/yosemite/mouse
/yosemite/odom
/yosemite/robot_description
/yosemite/robot_state/transition_event
/yosemite/scan
/yosemite/slip_status
/yosemite/static_transform/transition_event
/yosemite/stereo/depth
/yosemite/stop_status
/yosemite/tf
/yosemite/tf_static
/yosemite/wheel_status
/yosemite/wheel_ticks
/yosemite/wheel_vels

roni-kreinin, 2023-05-01:
Because you namespaced the robot you will have to also namespace the `teleop_twist_keyboard` node and the dock/undock actions.

Try running `teleop_twist_keyboard` like this: 

`ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -r __ns:=/yosemite`

To use the dock/undock actions, try:

`ros2 action send_goal /yosemite/dock irobot_create_msgs/action/Dock {}`
`ros2 action send_goal /yosemite/undock irobot_create_msgs/action/Undock {}`

tianmuw, 2023-05-01:
Thank you. But sometimes there is no odom topic, what is this issue? I connected the create3 to the same wifi and source the setup.bash file.

tianmuw, 2023-05-01:
Also, when there is the odom topic, I tried to use the dock/undock action, there is still showing that "Waiting for an action server to become available".

closed 2023-05-23 by tianmuw