turtlebot/turtlebot4 #155: joystick LED turns on red light.

rhyeonwoo, 2023-04-28:
ubuntu 20.04  ros galactic

I took out the joystick to check the robot for the first time, but there was no battery, so I connected the charger.
But as soon as I plugged in the joystick, I saw a red LED as shown in the picture.
As far as I know, the charging light flashes orange.

I charged it for two hours and disconnected the cable, but the red light is still on.
The red LED will remain on and not off until the joystick is discharged.
The pairing works and the operation works normally.
The issue is that if the red LED doesn't turn off, there is always a possibility of discharge.
Is there a way to solve this issue?

roni-kreinin, 2023-04-28:
Can you try pushing the reset button on the back of the controller?

rhyeonwoo, 2023-05-03:
I tried pressing the reset button in several ways, but there was no change.
Is there any action that needs to be taken after pressing the reset button?

roni-kreinin, 2023-05-03:
Try pressing and holding the home button for a few seconds. That should power it off.

If the issue persists you can contact support@clearpathrobotics.com and we will send you a replacement.

rhyeonwoo, 2023-05-04:
If you press the home button for more than a few seconds, it won't turn off, and you just have to wait until it's discharged.
I'll make an inquiry. Thank you.

closed 2023-05-22 by rhyeonwoo