turtlebot/turtlebot4 #156: Turtlebot's LCD doesn't come out

rhyeonwoo, 2023-04-28:
- OS: ( Ubuntu 20.04)
 - ROS Distro: (Galactic)

Turtlebot's LCD doesn't come out and joysticks don't connect.
Raspberry Pi and Create PC Wi-Fi are connected.
If I look at the topic, I can't see it all.  I compared it with other Turtle Bot products.
I think the Raspberry Pi has a light on, so the power is on.

I think mcu is the problem, is there any other problem?
[image]

rhyeonwoo, 2023-05-03:
"Connection timed out"  I am receiving an error message and cannot establish a connection.

roni-kreinin, 2023-05-03:
Are you able to SSH over ethernet as instructed [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi)?

rhyeonwoo, 2023-05-04:
No, "No route to host" an message is displayed and it is not connecting.

closed 2023-05-17 by rhyeonwoo