turtlebot/turtlebot4 #159: SLAM with namespaced Robots

ipa-rar, 2023-05-03:
**Describe the the feature you would like**
Documentation on using namespace robots with the turtlebot4_navigation. This is not found in the ``multiple_robots.md`` and its very critical for multirobot users.

**Other notes**
Add anything else you thing is important.

roni-kreinin, 2023-05-03:
We will add that. You can use the `namespace` launch argument to launch navigation for specific robots.

`ros2 launch turtlebot4_navigation slam.launch.py namespace:=/robot1`

ipa-rar, 2023-05-03:
@roni-kreinin What other changes has to be made to the [parameters](https://github.com/turtlebot/turtlebot4/tree/humble/turtlebot4_navigation/config) inorder to get it running. In my case I have frame_prefix to the links.

roni-kreinin, 2023-05-03:
Are you trying to generate a single map using multiple robots? Or a separate map for each robot?

ipa-rar, 2023-05-03:
A single map that can be shared by multiple robots once mapping is complete. But each robot will be using the same map for navigation within a closed arena.

roni-kreinin, 2023-05-03:
We don't yet have any examples of how to run this, but that does seem to be the correct approach. If you are able to get this running then feel free to make a PR to https://github.com/turtlebot/turtlebot4_tutorials and we can add a tutorial.

ipa-hsd, 2023-05-04:
Wouldn't things be easier if the turtlebot Xacros include `prefix` parameter for links and joints names? That way each turtlebot can be launched with its specific prefix.