turtlebot/turtlebot4 #161: I cannot subscribe to iCreate ir_intensity

nidia-bucarelli, 2023-05-04:
Hello,
I am having issues with the ir_intensity topic. How I can subscribe to the topic? Below is what I have tried with no success. 
Any feedback?

1. I had to change the RMW implementation to use fastrtps to be able to see the icreate messages in the command line (using ros2 topic list)
2.	I was able to get the ir_intensity message in the command line using the topic /base/ir_intensity
```
$ros2 topic echo /base/ir_intensity 

header:
  stamp:
    sec: 1683178277
    nanosec: 978064616
  frame_id: base_link
readings:
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_side_left
  value: 135
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_left
  value: 21
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_front_left
  value: 7
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_front_center_left
  value: 5
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_front_center_right
  value: 2
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_front_right
  value: 5
- header:
    stamp:
      sec: 1683178277
      nanosec: 978064616
    frame_id: ir_intensity_right
  value: 0
```
3.	I wrote python code to subscribe to the same topic (/base/ir_intensity), but I do not get any messages in the callback function.
4.	I tried adding a sleep for 1 second before subscribing as suggested by some people so I can get access to all the topics (using the rclpy method self.get_topic_names_and_types())
5.	The code now shows all the topics but does not subscribe to the icreate topic /base/ir_intensity).

The code:
```
import rclpy 
from rclpy.node import Node

from irobot_create_msgs.msg import IrIntensityVector
from rclpy.qos import ReliabilityPolicy
import time
[... 130 lines cut ...]
('/imu', ['sensor_msgs/msg/Imu'])
('/interface_buttons', ['irobot_create_msgs/msg/InterfaceButtons'])
('/ip', ['std_msgs/msg/String'])
('/joint_states', ['sensor_msgs/msg/JointState'])
('/joy', ['sensor_msgs/msg/Joy'])
('/joy/set_feedback', ['sensor_msgs/msg/JoyFeedbackArray'])
('/led_animation/_action/feedback', ['irobot_create_msgs/action/LedAnimation_FeedbackMessage'])
('/led_animation/_action/status', ['action_msgs/msg/GoalStatusArray'])
('/mouse', ['irobot_create_msgs/msg/Mouse'])
('/parameter_events', ['rcl_interfaces/msg/ParameterEvent'])
('/robot_description', ['std_msgs/msg/String'])
('/rosout', ['rcl_interfaces/msg/Log'])
('/scan', ['sensor_msgs/msg/LaserScan'])
('/stereo/camera_info', ['sensor_msgs/msg/CameraInfo'])
('/stereo/depth', ['sensor_msgs/msg/Image'])
('/tf', ['tf2_msgs/msg/TFMessage'])
('/tf_static', ['tf2_msgs/msg/TFMessage'])
('/undock/_action/feedback', ['irobot_create_msgs/action/Undock_FeedbackMessage'])
('/undock/_action/status', ['action_msgs/msg/GoalStatusArray'])
('/wall_follow/_action/feedback', ['irobot_create_msgs/action/WallFollow_FeedbackMessage'])
('/wall_follow/_action/status', ['action_msgs/msg/GoalStatusArray'])
('/wheel_status', ['irobot_create_msgs/msg/WheelStatus'])
subscriber started

```

roni-kreinin, 2023-05-15:
@nidia-bucarelli You will need to import `from rclpy.qos import qos_profile_sensor_data` and replace `ReliabilityPolicy.SYSTEM_DEFAULT` with `qos_profile_sensor_data` in your subscriber.

nidia-bucarelli, 2023-05-15:
Thank you, Roni. 
It seems to be working now.

closed 2023-05-15 by roni-kreinin