turtlebot/turtlebot4 #163: Cant connect to Rpi4 , how do a factory reset or some reset for default values

sashag1991, 2023-05-09:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: Turtlebot4 already installed image on Rpi4

 **Expected behaviour**
Connect to Rpi4 with LAN without issue

 **Actual behaviour**
 Cant connect Rpi4 via LAN cable

**To Reproduce**
Provide the steps to reproduce:
1. Connected first time with LAN to Rpi4 
2. Did some changes in netplan file 
3. sudo reboot
4. Couldn't connect again to same IP LAN address

**Other notes**
Is there a way to do hard reset or factory reset to the platform

ipa-rar, 2023-05-09:
@sashag1991. 
 You can modify the netplan manually but in my experience you should not do that on Turtlebot4. I assume you have followed the [similar](https://github.com/turtlebot/turtlebot4_setup#manually-configure-wi-fi) steps to configure your LAN. I will not advice to try out this method. 
 
 **To hard reset** - You have only one option here:  Download the new image and reflash the sd card.
 https://github.com/turtlebot/turtlebot4_setup/issues/5#issuecomment-1497772016
 
 After this process only connect to the tb4 via WiFi and then use turtlebot4-setup to change your configurations.

ipa-rar, 2023-05-09:
On the first boot, the Raspberry Pi will enter Access Point (AP) mode which will allow you to connect to it over Wi-Fi. On your PC, connect to the Turtlebot4 Wi-Fi network. The password is also Turtlebot4.

Once connected, you can SSH into the Raspberry Pi to configure its Wi-Fi. 

[You can follow this for Basic setup and this works. ](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html).
**Please do not manually edit the yaml files yourself again.**

sashag1991, 2023-05-09:
Seems that https://github.com/turtlebot/turtlebot4_setup/issues/5#issuecomment-1497772016 solved my issues

ipa-rar, 2023-05-12:
@sashag1991 Can you please close this issue if it solved.

roni-kreinin, 2023-05-15:
@ipa-rar Thanks for helping them!

closed 2023-05-15 by roni-kreinin

mocaplabiitj, 2024-09-12:
After Configuring the Turtlebot4 to connect to Wi-Fi, I can not connect to the robot (with correct credentials an DHCP - False). The IP address line shows _UNKNOWN_ message with battery level _0_ (even after charging for a long time).

I tried to connect through the ethernet but couldn't succeed as the terminal shows the time out message.

How can open the robot to access the HDMI port or the SD card for hard reset of the robot?