turtlebot/turtlebot4 #165: How run mapping/navigation launches on stationary desktop with turtlebot4

sashag1991, 2023-05-11:
**Please provide the following information:**
 - OS:  Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed:

I installed all packages for ROS2 + turtulebot4 on stationary PC.
How do I connect turtlebot4 with all the sensors to to stationary PC to run mapping/navigation?
Currently turtlebot4 with Create 3 have both IP on same network as stationary PC

sashag1991, 2023-05-11:
I followed this tutorial: https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html
I have wifi connection between remote PC <-> Create 3 and  remote PC <-> Rpi4

ipa-rar, 2023-05-12:
You should be able to  run mapping/navigation on stationary desktop with turtlebot4 just by following this. 
https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html#navigation
`slam.launch.py `and `nav2.launch.py` should be launched on the PC

sashag1991, 2023-05-14:
I follow the instructions with these launches:
1. ros2 launch turtlebot4_navigation slam.launch.py
2. ros2 launch turtlebot4_navigation nav2.launch.py
3. ros2 launch turtlebot4_viz view_robot.launch.py

 controller_server-1] [INFO] [1684053542.503579202] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

echo to topic /scan in remote PC , see nothing :
[image]

sashag1991, 2023-05-14:
Additionally cant see create 3 topics as battery_status and wheel_status in remote PC

ipa-rar, 2023-05-15:
> controller_server-1] [INFO] [1684053542.503579202] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

I had the same issue and this is the absence of `base_link -> odom` transform. Few things which I have noted are
1. I think this transform will be provided when we launch the slam but this does not work as expected everytime (#167 )
2. Create3 is running out of processing or memory (#147 )
3. There might be a time sync issue which might be causing all these problems (#166 )(#117 )(#62 )(#143 )

I still have not got my tb4 working even after >4 months of work but I would keep a close watch on these issues to fix it. But in my opinion TB4 is not completely reliable and unstable. I will wait for some major updates to roll out.

sashag1991, 2023-05-15:
But in the first place I don't see any topic in remote PC. How should I test the topic transfer between TB4 standard (RPI4) with remote PC (desktop with Jammy 22.04)

ipa-rar, 2023-05-15:
@sashag1991 
1. Did you set the same Domain ID as that of TB4 in the remote PC
2. Did you also set the DDS to be same as TB4

sashag1991, 2023-05-15:
Yes, I use fastDDS and domain ID as 0 (by default)

ipa-rar, 2023-05-15:
You should also try running the slam.launch on the tb4 and then see

roni-kreinin, 2023-05-15:
@sashag1991 Can you try using the [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)?

sashag1991, 2023-05-16:
How [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) should help ? I still cant see TB4 topics

roni-kreinin, 2023-05-16:
Discovery server allows you to disable the Wi-Fi on the Create 3. The data will only be sent over the USB-C cable and forwarded through the RPi Wi-Fi. 

The issue you are having could be caused by the Create 3's Wi-Fi connection, so the discovery server may fix that.

sashag1991, 2023-05-17:
[Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) indeed helped :).

I cant understand which launched I should run on TB4 and remote PC to generate a new map?

roni-kreinin, 2023-05-17:
Instructions for generating a map can be found [here](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html). Basically you will want to run SLAM on either the TB4 or your PC. Both should work, but generally running it on a PC will perform better. Then you can run RViz to visualise the map.

sashag1991, 2023-05-23:
I launched the following on the remote PC:

ros2 launch turtlebot4_navigation slam.launch.py sync:=false

[image]

ros2 launch turtlebot4_viz view_robot.launch.py

[image]

In RVIZ I see nothing, maybe I miss something ?

[image]

roni-kreinin, 2023-05-23:
Are you able to see data on the `/scan` topic from your PC?

sashag1991, 2023-05-24:
After power cycle it see now the topic and also in RVIZ, but why I cant see the map that it should create ? 

[image]

sashag1991, 2023-05-24:
When I changed the fixed frame to map, I got gloabal status: Error:

[image]

sashag1991, 2023-05-24:
I also noticed that not all topics exist, I'm missing TB4 topics like : /dock , /wheel_ticks , /wheel_vels, /stereo/camera_info, 
/stereo/depth and etc'.
This is the full topic list I should have (including TB4): 
/battery_state
/clicked_point
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/downsampled_costmap
/downsampled_costmap_updates
/global_costmap/costmap
/global_costmap/costmap_updates
/global_costmap/voxel_marked_cloud
/hazard_detection
/imu
/initialpose
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/kidnap_status
/local_costmap/costmap
/local_costmap/costmap_updates
/local_costmap/published_footprint
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_metadata
/map_updates
/mobile_base/sensors/bumper_pointcloud
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/particle_cloud
/plan
/robot_description
/robot_state/transition_event
/rosout
/scan
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/waypoints
/wheel_status
/wheel_ticks
/wheel_vels

roni-kreinin, 2023-05-25:
Can you paste the output of `ip route` from your PC?

sashag1991, 2023-05-31:
ip route on remote PC :
[image]

ip route on turtlebot4
[image]

sashag1991, 2023-05-31:
192.168.1.45 is the turtlebot4 IP

roni-kreinin, 2023-06-20:
@sashag1991 Sorry for the delay.

Could you share the following:
- The application configuration of the Create 3 from the "Application" tab in the webserver
- The contents of `/etc/turtlebot4/setup.bash` from the RPi4
- The contents of `/etc/turtlebot4_discovery/setup.bash` from your PC
- The contents of `/etc/utrtlebot4_discovery/fastdds_discovery_super_client.xml` from your PC

sashag1991, 2023-07-04:
@roni-kreinin, Sorry for my delay.

Here are the results:
1. Create 3 from the "Application" tab in the webserver:

[image]

2. contents of /etc/turtlebot4/setup.bash :

export CYCLONEDDS_URI=/etc/turtlebot4/cyclonedds_rpi.xml
export FASTRTPS_DEFAULT_PROFILES_FILE=/etc/turtlebot4/fastdds_discovery_super_client.xml
export ROBOT_NAMESPACE=
export ROS_DOMAIN_ID=0
export ROS_DISCOVERY_SERVER=127.0.0.1:11811
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export TURTLEBOT4_DIAGNOSTICS=1
export WORKSPACE_SETUP=/opt/ros/humble/setup.bash

3. content of /etc/turtlebot4_discovery/setup.bash :

source /opt/ros/humble/setup.bash
export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
export FASTRTPS_DEFAULT_PROFILES_FILE=/etc/turtlebot4_discovery/fastdds_discovery_super_client.xml
export ROS_DISCOVERY_SERVER=192.168.1.45:11811
export ROS_DOMAIN_ID=0

4. content of  /etc/utrtlebot4_discovery/fastdds_discovery_super_client.xml

   <?xml version="1.0" encoding="UTF-8" ?>
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
        <participant profile_name="super_client_profile" is_default_profile="true">
            <rtps>
                <builtin>
                    <discovery_config>
                        <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                        <discoveryServersList>
                            <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                                <metatrafficUnicastLocatorList>
                                    <locator>
                                        <udpv4>
                                            <address>192.168.1.45</address>
                                            <port>11811</port>
                                        </udpv4>
                                    </locator>
                                </metatrafficUnicastLocatorList>
                            </RemoteServer>
                        </discoveryServersList>
                    </discovery_config>
                </builtin>
            </rtps>
        </participant>
    </profiles>
</dds>

hilary-luo, 2023-09-21:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-21 by hilary-luo

maalon, 2023-12-27:
i have the same problem

maalon, 2023-12-27:
ros2 topic list from the raspberry pi :
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
ros2 topic list from the pc:
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/dock_status
/hazard_detection
/imu
/interface_buttons
/ir_intensity
/ir_opcode
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_state/transition_event
/rosout
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

maalon, 2023-12-27:
when i run ros2 launch turtlebot4_navigation nav2.launch.py that what i got 
[planner_server-3] [INFO] [1703677499.104104907] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist

hilary-luo, 2023-12-27:
@maalon Please open your own ticket, rather than commenting on a closed ticket with your own issue. That being said, launching nav2 on its own means that you are launching navigation but that does not provide a map, only the ability to navigate within a map. You need to also launch either slam to create a new map or localization to navigate within a saved map. Please see the information and tutorials on the website: [Navigation and SLAM vs Localization](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html) and [Turtlebot4 Navigator](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/turtlebot4_navigator.html)