turtlebot/turtlebot4 #166: SLAM map not working in hot spot mode for TB4 Lite we got in May, 2023

idrobot, 2023-05-11:
**Please provide the following information:**
I have a new arrival TB4 Lite in early May.
In PC:
- OS: Ubuntu 20.04
- ROS Distro: Galactic
In robot:
- turtlebot4_galactic_0.1.3 + Create3-G.4.4

 **Expected behaviour**
If I set TurtleBot4 Lite in hot spot mode(Rpi4 in ap mode)，in PC, it should show SLAM map in Rviz。

 **Actual behaviour**
TurtleBot 4 Lite only show blank SLAM map

**To Reproduce**
1.	In PC, ssh to Rpi4 Wifi to change hot spot mode configuration
ssh ubuntu@10.42.0.1
sudo nano /etc/netplan/50-cloud-init.yaml

[image]

(we mark off 5Ghz band and reconfigure network)
sudo netplan apply

2.	In PC, open iRobot page, find TB4 Wifi ID and connect Wifi
http://10.42.0.1:8080/wifi

3.	In PC, check TB4 topics and launch SLAM, Rviz
ros2 launch turtlebot4_viz view_diagnostics.launch.py
ros2 launch turtlebot4_navigation slam_sync.launch.py
ros2 launch turtlebot4_viz view_robot.launch.py

4.	It only show blank SLAM

[image]

**Other notes**

1.	We have old TB4 Lite which we got in March, 2023. That TB4 Lite run
(turtlebot4_galactic_0.1.2+Create3-G.2.2 or turtlebot4_galactic_0.1.3＋Create3-G.3.1)

I follow above same procedures, normal SLAM map is able to show in hot spot mode, like following:

[image]

2.	If I upgrade old TB4 Lite(March one) to new TB4 configuration: turtlebot4_galactic_0.1.3 + Create3-G.4.4
OK, normal SLAM map is also showing in hot spot mode.

3.	If I downgrade new TB4 Lite(early May one) from (turtlebot4_galactic_0.1.3 + Create3-G.4.4) to old configuration (turtlebot4_galactic_0.1.2+Create3-G.2.2 or turtlebot4_galactic_0.1.3＋Create3-G.3.1)
Not working. Blank SLAM.

I think something wrong in early May vesion of TB4 Lite.

roni-kreinin, 2023-05-15:
@idrobot When you run into the issue with the map not showing, are you able to echo the `/odom` topic from your PC? It sounds like your PC might not be seeing the odom data.

idrobot, 2023-05-22:
> @idrobot When you run into the issue with the map not showing, are you able to echo the `/odom` topic from your PC? It sounds like your PC might not be seeing the odom data.

Hi roni, please see my replies
1. in PC, ros2 topic list, there is /odom topic
[image]

2. and then ros2 topic echo /odom, data received
[image]

3. I think this problem is not related with odom. I check https://iroboteducation.github.io/create3_docs/setup/compute-ntp/ and found rpi can not sync time between NTP server and create 3. Once we connect to internet, time will be sync, and then SLAM map will show. This problem should be related with time sync.

closed 2023-09-21 by hilary-luo