turtlebot/turtlebot4 #167: SLAM without Obstacle avoidance

ipa-rar, 2023-05-12:
**Please provide the following information:**

>  - Ubuntu 202.04
>  - ROS Distro: Humble
>  - installed:
>  - Create3 version: H 1.2
> - Simple discovery as I plan to use this on a multi-robot setup(#159 )

 **Expected behaviour**

> SLAM with obstacle avoidance during the navigation

 **Actual behaviour**

> SLAM works, Navigation works but obstacles are not avoided on the way.(Tested on real robot)

**To Reproduce**
Provide the steps to reproduce:
1. `ros2 launch turtlebot4_navigation localization.launch.py map:=/path/to/map.yaml`
2. `ros2 launch turtlebot4_navigation nav2.launch.py`
3. `ros2 launch turtlebot4_viz view_robot.launch.py`

- This is the result after 5 minutes 
[image]

**Other notes**

( #141 ) is almost similar issue but did not help either.

**Few things I have noticed with my issue** 

> For the Navigation part to be active it takes very long time and for me it took more than 5 minutes for the system to be active. I am running on a 8th gen i7 with 16 GB RAM. 

1. This is the tf_tree when the complete system is up and running. Here the tree is connected and looks fine.
[image]

2. And gave a nav2 goal and the robot started moving to the target and reaches the place. Again tested with a new target with an obstacle on its way. This time the robot hits the obstacle and does not even detect anything in rviz. Repeating this several times, it was found that the laser scan data disappear in between and the error looks like this
[image]

3. The tf tree looks like this at the same time
[image]

> odom --> base_link is broken and I could not find a solution for this

This looks similar to (#158 ) but is still unresolved.

**My theory**
- broken odom--> base_link affects the obstacle avoidance as the robot looses the /Scan messages at this point. Without the /Scan data the robot is navigating blind. 

Anyone else who figured out a reason for this behavior or a solution to overcome this behavior please suggest. 

**Fun part**
 odom--> base_link  tree link connects and disconnects which means something related to this is really unstable.

roni-kreinin, 2023-05-15:
@ipa-rar Can you check the CPU usage of the Create 3 by looking at the logs in the webserver? This might be related to the bug where the Create 3 is hitting 100% CPU usage which causes delays in publishing the odom->base_link transform.

ipa-rar, 2023-05-17:
> Create 3 is hitting 100% CPU usage

Yes, The logs show that the CPU is at 100% and I also thought this could be the reason. However this is not expected and the tb4 should be able to perform SLAM and navigation.

Jayden-F, 2023-05-19:
@ipa-rar  @roni-kreinin The create3 is consistently the single most frequent point of failure for the TB4. Which would be ok if it was open source, however being closed source it constantly gets in the way of progress.

hilary-luo, 2023-11-16:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process and summarize where you are at.

closed 2023-11-16 by hilary-luo