turtlebot/turtlebot4 #169: Discovery Server,

ddwy, 2023-05-16:
Please provide the following information:

- OS: Ubuntu 22.04
- ROS Distro: Humble
- ROS 2 networking: Discovery Server

Configuring turbobot4 as Discovery Server,then I can view the topic, but I cannot view the data of the topic,such as /battery_state

roni-kreinin, 2023-05-16:
Can you ping `192.168.186.2` from the RPI?

ddwy, 2023-05-17:
> Can you ping `192.168.186.2` from the RPI?

No,i can't.

ddwy, 2023-05-18:
> Can you paste the output of `ip a` on the RPi? And also the contents of the following files:
> 
> * `/etc/netplan/40-ethernets.yaml`
> * `/boot/firmware/cmdline.txt`
> * `/boot/firmware/config.txt`

[image]

roni-kreinin, 2023-06-20:
Closing due to inactivity. Feel free to reopen if the issue persists.

closed 2023-06-20 by roni-kreinin

vatsuak, 2023-11-01:
Hi, 
I am getting the same issue.

I can list the topics but cannot see the data on the host machine

1. I can ping 192.168.186.2 from the pi
2. contents of /etc/netplan/40-ethernets.yaml

[image]

3. contents of /boot/firmware/cmdline.txt
[image]

4. contents of /boot/firmware/config.txt
[image]

The switch is set to USB on the connector. 

Also, this started to happen after I rebooted the pi after installing the discovery server. It was working ok before the reboot.