turtlebot/turtlebot4 #170: Not able to save map in humble

etigafy, 2023-05-17:
**Using:** 
- OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed

 **Expected behaviour**
SaveMap service saves the map in the current location

 **Actual behaviour**
```
ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'my_map'"
requester: making request: slam_toolbox.srv.SaveMap_Request(name=std_msgs.msg.String(data='my_map'))

response:
slam_toolbox.srv.SaveMap_Response(result=255)
```

Service responds with undefined failure.

**To Reproduce**
Provide the steps to reproduce:
1. ros2 launch turtlebot4_navigation slam.launch.py
2. ros2 launch turtlebot4_viz view_robot.launch.py
3. ros2 service call /slam_toolbox/save_map slam_toolbox/srv/SaveMap "name:
  data: 'my_map'"

roni-kreinin, 2023-05-25:
Do you have another map with the name `my_map` in the directory you call this from?

etigafy, 2023-05-30:
No. I tried it again this morning the behavior is still the same...

roni-kreinin, 2023-05-30:
The result code (255) isn't very helpful for determining what is causing this issue. I'm going to close this issue and I would recommend re-opening it at https://github.com/SteveMacenski/slam_toolbox. I think those devs will have better insight into what could cause this.

closed 2023-05-30 by roni-kreinin

ACdu22330, 2023-06-02:
Hello, I have the same trouble as @etigafy  did you succeed to fix this problem?