turtlebot/turtlebot4 #171: The camera's connectivity varies depending on the USB port.

gamsaon, 2023-05-22:
OS: Ubuntu 22.04
ROS Distro: Humble
turtlebot4_Humble+ Create3-H_1.2

Expected behaviour
The camera needs to work properly

Actual behaviour
If I connect the camera USB to USB 2.0, there is no problem. However, it does not work when I connect the camera to USB 3.0.

To Reproduce
1. When I connected the camera to USB 2.0.
[image]

As you can see from the photo, when I ran dmesg, it indicates that the connection did not get disconnected. Additionally, camera-related values are displayed correctly.

3. When I connected the camera to USB 3.0
[image]

As shown in the photo, when I ran dmesg, it indicated that the USB got disconnected. Additionally, in the camera topic list, the camera is listed but no values are displayed. Furthermore, the camera works fine until it is connected to the PC, but as soon as it is connected, the camera connection gets interrupted. 

 Is it okay to use the camera on USB 2.0? What are the possible solutions to this issue?

roni-kreinin, 2023-05-23:
Can you share the logs from the `turtlebot4` service when you are using USB 3.0?

`sudo journalctl -u turtlebot4 -r`

gamsaon, 2023-05-24:
May 24 02:23:15 ubuntu turtlebot4-start[988]: [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oakd' in container '/oakd_container'
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.718442955] [oakd]: Camera ready!
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.353593053] [oakd]: Finished setting up pipeline.
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.299845955] [oakd]: Pipeline type: RGB
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.294922269] [oakd]: Device type: OAK-D-PRO-AF
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.272513090] [oakd]: USB SPEED: SUPER
May 24 02:23:15 ubuntu turtlebot4-start[988]: [component_container-6] [INFO] [1684894995.267831923] [oakd]: Camera with MXID: 18443010B1AE361300 and Name: 1.1.1 connected!
May 24 02:23:12 ubuntu turtlebot4-start[988]: [rplidar_composition-5] [INFO] [1684894992.935454768] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2, flip_x_axis 0
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.749033742] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.749013539] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Mouse'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748957818] [GenericAnalyzer]: Analyzer 'Mouse' matches 'turtlebot4_diagnostics: /mouse topic status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748715265] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748684210] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'IMU'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748517009] [GenericAnalyzer]: Analyzer 'IMU' matches 'turtlebot4_diagnostics: /imu topic status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748382399] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Camera'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748345826] [GenericAnalyzer]: Analyzer 'Camera' matches 'turtlebot4_diagnostics: /color/preview/image topic status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748225346] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Camera'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748186828] [GenericAnalyzer]: Analyzer 'Camera' matches 'turtlebot4_diagnostics: /stereo/depth topic status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.748035015] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer 'Lidar'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747975701] [GenericAnalyzer]: Analyzer 'Lidar' matches 'turtlebot4_diagnostics: /scan topic status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747696446] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747677372] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Hazards'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747647799] [GenericAnalyzer]: Analyzer 'Hazards' matches 'turtlebot4_diagnostics: Hazard Detections'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747531022] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747509634] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Dock'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747478949] [GenericAnalyzer]: Analyzer 'Dock' matches 'turtlebot4_diagnostics: Dock Status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747364025] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747346043] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Wheels'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747305544] [GenericAnalyzer]: Analyzer 'Wheels' matches 'turtlebot4_diagnostics: Wheel Status'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.747049881] [AnalyzerGroup]: Group 'Turtlebot4' has a match with my analyzer '/Turtlebot4/Create3'.
May 24 02:23:12 ubuntu turtlebot4-start[988]: [aggregator_node-10] [INFO] [1684894992.746973326] [AnalyzerGroup]: Group '/Turtlebot4/Create3' has a match with my analyzer 'Battery'.
[... cut ...]
[   37.809292] usb 2-1: SerialNumber: 18443010B1AE361300
[   42.975995] usb 2-1: USB disconnect, device number 3
[   45.199974] usb 1-1.1: new high-speed USB device number 6 using xhci_hcd
[   45.304706] usb 1-1.1: New USB device found, idVendor=03e7, idProduct=2485, bcdDevice= 0.01
[   45.304730] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   45.304738] usb 1-1.1: Product: Movidius MyriadX
[   45.304745] usb 1-1.1: Manufacturer: Movidius Ltd.
[   45.304751] usb 1-1.1: SerialNumber: 03e72485

roni-kreinin, 2023-05-25:
Does this happen with both USB 3 ports? Also can you try this again with the RPLIDAR and any other USB devices disconnected?

kkksion, 2023-05-30:
The same issue occurs where the camera does not turn on in two USB 3.0 ports. Furthermore, as you suggested, when I disconnected the RPLIDAR and ran it on a USB 3.0 port, a new problem arose. The camera works fine until the robot finishes booting, but once the booting process is complete, the camera no longer works

roni-kreinin, 2023-05-30:
What do you mean by `The camera works fine until the robot finishes booting`? Are you getting this from dmesg?

Also, do you have a TurtleBot 4 lite or standard?

gamsaon, 2023-05-31:
I am currently using Turtlebot4 standard. Today, I double-checked, and it seems that what I mentioned earlier was incorrect. When I tried again after disconnecting the RPLIDAR and other USB devices, the camera connected to USB 3.0 did not work. Among many robots, only this robot seems to be unable to use the camera when it is connected to USB 3.0.

closed 2023-09-21 by hilary-luo