turtlebot/turtlebot4 #172: ROS topics not shown on remote PC

huayil, 2023-05-22:
PC OS: Ubuntu 22.04
PC ROS version: Humble
Create® 3 firmware version: H.1.2
Create® 3 has been factory reset
Raspberry Pi: installed turtlebot4_lite_humble_1.0.0 then sudo apt update; sudo apt upgrade
Network configuration: Discovery Server

Running `ip route` on the PC gives me `192.168.186.0/24 via <My Raspberry Pi IP> dev wlp0s20f3` where `<My Raspberry Pi IP>` matches my Raspberry Pi's IP address.

Running `ros2 topic list` on the remote PC does not return the Raspberry Pi or Create® 3 topics.

Could you please provide some troubleshooting methods? Thanks!

roni-kreinin, 2023-05-23:
Can you try running `ros2 daemon stop; ros2 daemon start` before calling `ros2 topic list`?

huayil, 2023-05-23:
> 

I have run that before calling `ros2 topic list` but the issue persists. I also made sure I call `ros2 topic list` twice.

Further information: 
- There is response when I `ping` the Raspberry Pi IP. 
- I tried `ros2 run demo_nodes_cpp talker` after `ssh` into the Raspberry Pi, and `ros2 run demo_nodes_cpp listener` on the PC. The listener node has no response.
- After following the [tutorial](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#discovery-server) to set up the Discovery Server, when running the talker and listener nodes on the PC, the listener node has no response. If I comment out `export FASTRTPS_DEFAULT_PROFILES_FILE=/etc/turtlebot4_discovery/fastdds_discovery_super_client.xml` and `export ROS_DISCOVERY_SERVER=<My Raspberry Pi IP>` in `/etc/turtlebot4_discovery/setup.bash`, then the listener node has response.

roni-kreinin, 2023-05-25:
Can you share the contents of `/etc/turtlebot4/setup.bash` on the RPi4? It sounds like the RPi might not be configured for the Discovery Server which is why your listener node works when you comment out the discovery server exports.

huayil, 2023-05-26:
I disabled the firewall and the problem was resolved so the cause is related to the firewall configuration. 

@roni-kreinin  Thanks for your suggestions.

closed 2023-05-26 by huayil

jblevins32, 2024-06-22:
Can you let us know what the solution was?

stergiosba, 2024-11-01:
@jblevins32 He just did:
```bash
sudo ufw disable
```
To what degree this qualifies as a solution is....questionable.