turtlebot/turtlebot4 #173: TF etc really flaky TurtleBot Lite

CycleMark, 2023-05-22:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from installed:

I've set up the network as per https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html [I don't want the Create3 connected to the network - that seems to make matters worse]

Starting any/all of:

ros2 launch turtlebot4_bringup lite.launch.py
ros2 launch rplidar_ros rplidar.launch.py

Running:

ros2 run tf2_tools view_frames

On either the RPi or the PC (I have set up the PC as per inx above) calling:

ros2 run tf2_tools view_frames

It intermittent - Sometimes the TF tree most of the time not.

I've run through:

https://turtlebot.github.io/turtlebot4-user-manual/overview/

And can't see anything I've missed.

Could you please advise where I can check next. 

Thanks

Mark

ipa-rar, 2023-05-25:
@CycleMark Did you set namespace to your robot. If thats the case, then ``ros2 run tf2_tools view_frames`` will not show any trees. Try this [node](https://github.com/ipa-rar/tf_relay)

CycleMark, 2023-05-25:
@ipa-rar  - Thanks but that isn't the issue. The namespace is set correctly - plus this is only one robot which sometime does and then doesn't show TF - so intermittent

roni-kreinin, 2023-05-25:
@CycleMark Are you stopping the `turtlebot4` service before calling `ros2 launch turtlebot4_bringup lite.launch.py`? The service runs `lite.launch.py` in the background so you would be launching duplicate nodes.

You can check if the service is running with `sudo systemctl status turtlebot4`, and view the full logs with `sudo journalctl -u turtlebot4 -r`.

closed 2023-09-21 by hilary-luo