turtlebot/turtlebot4 #174: Turtlebot4 after update/upgrade not booting correctly.

Cur1ousKea, 2023-05-23:
@roni-kreinin I reinstalled the humble image a third time, after the turtlebot didnt boot properly after an update/upgrade. Now it starts up with the 5 green leds flashing as it should be.
A question i would have: The controller when being pressed slow forward for once it turns in a circle moves a bit forward and when it hits sth it turns and move on for 5 sec or so... When i installed it the first time it worked properly, not mentioning the battery issues.

Furthermore when trying to update and upgrade - after the reboot the lcd screen doesnt show the ip adress any more and only the mtr and power led are green. the weird thing is that i can access the raspi via wifi and execute everything as if it was normal... Tried starting the services which were stopped and it worked but the status leds are still off. when i want to start diagnostics it doesnt show anything anymore. the topic list is smaller but the same as on rpi:
this is the status:
sudo service turtlebot4 status
× turtlebot4.service - "bringup turtlebot4"
Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Sun 2023-05-21 21:29:38 UTC; 3s ago
Process: 2190 ExecStart=/usr/sbin/turtlebot4-start (code=exited, status=1/FAILURE)
Main PID: 2190 (code=exited, status=1/FAILURE)
CPU: 8.981s
May 21 21:29:38 ubuntu turtlebot4-start[2249]: File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
May 21 21:29:38 ubuntu turtlebot4-start[2249]: while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
May 21 21:29:38 ubuntu turtlebot4-start[2249]: File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
May 21 21:29:38 ubuntu turtlebot4-start[2249]: while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
May 21 21:29:38 ubuntu turtlebot4-start[2249]: File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
May 21 21:29:38 ubuntu turtlebot4-start[2249]: with self.handle:
May 21 21:29:38 ubuntu turtlebot4-start[2249]: rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
May 21 21:29:38 ubuntu systemd[1]: turtlebot4.service: Main process exited, code=exited, status=1/FAILURE
May 21 21:29:38 ubuntu systemd[1]: turtlebot4.service: Failed with result 'exit-code'.
May 21 21:29:38 ubuntu systemd[1]: turtlebot4.service: Consumed 8.981s CPU time.

And the last remark i would have is the camera. i wanted to start the depthai_demo.py but it says that it doesnt detect the necessary camera. By having the Turtlebot4 this shouldnt be the case i assume. Ich checked with lsusb and it is shown there. when trying to start it in rviz2, it says Stereo is not supported. I tried starting the service and im getting the following error
[Cameralog.txt](https://github.com/turtlebot/turtlebot4/files/11548676/Cameralog.txt)

roni-kreinin, 2023-05-25:
Can you send the full logs from the last boot using `sudo journalctl -u turtlebot4 -r`?

Cur1ousKea, 2023-05-25:
Appended
[output.txt](https://github.com/turtlebot/turtlebot4/files/11569494/output.txt)

roni-kreinin, 2023-05-26:
I'm not seeing the errors from `sudo service turtlebot4 status` in the logs. What happens if you restart the service with `sudo systemctl restart turtlebot4`?

Cur1ousKea, 2023-06-01:
Well i tried it one more time. This time not restarting the services with the UI which shows the services who have to be restarted. I went on cancel and rebooted manually. Thats how it works for me.

closed 2023-06-01 by Cur1ousKea

llyf-s, 2023-06-14:
> Well i tried it one more time. This time not restarting the services with the UI which shows the services who have to be restarted. I went on cancel and rebooted manually. Thats how it works for me.

Hi, can i check what are the steps you took for this? I am currently facing the same issue.

Pooja-KR5, 2024-12-26:
Hello, I have the same issue as well. Can I know what are the steps you used to resolve this issue?

Pooja-KR5, 2024-12-26:
> > Well i tried it one more time. This time not restarting the services with the UI which shows the services who have to be restarted. I went on cancel and rebooted manually. Thats how it works for me.
> 
> Hi, can i check what are the steps you took for this? I am currently facing the same issue. tried ctrl+c and reboot but LED still wouldnt turn on.

Hi, were you able to fix it?