turtlebot/turtlebot4 #175: Create 3 - Cannot connect to Wifi

Zakant, 2023-05-24:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed

 **Expected behaviour**
The Create3 platform can connect to a wifi access point after providing the SSID and psk.

 **Actual behaviour**
The Create3 fails to connect to the ap. This behaviour is regardless of the firmware version. As seen in the attached log file, the platform “failed to stat file network.ssid”. Occasionally, the platform manages to establish a connection - SSID and psk can therefore not be wrong - but loses it again very quickly. The wifi used is a standard “home” wifi (so not a cooperate one with special rules) providing both a 2,4 and 5 GHz connection under the same SSID. 

**To Reproduce**
Provide the steps to reproduce:
1. Factory reset the Create platform
2. Try to connect to a wifi
3. see the error in the logs and obtain no connection

Log file from the platform: [create3_logs.txt](https://github.com/turtlebot/turtlebot4/files/11550815/create3_logs.txt)

roni-kreinin, 2023-05-25:
@Zakant Can you try separating your network into 2.4GHz and 5GHz SSIDs and connecting to just the 2.4?

Zakant, 2023-05-26:
Surprisingly, it worked (the instructions only talk about the platform needing to be connected to a 2.4 GHz network, not that there may also be a 5 GHz with the same SSID...).
However, our IT is very unhappy to split the bands. Is this a known bug that will be fixed in a future update or is this by-design?

roni-kreinin, 2023-05-26:
The Create 3 is only able to connect to 2.4GHz because of hardware limitations. I would recommend that you switch to using the [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/networking.html#discovery-server) to avoid using the Create 3 Wi-Fi all together.

Zakant, 2023-05-26:
Sorry that I did not formulate my question clearly. There was a 2.4 GHz network at all times. The only difference is that once it had the same SSID as a 5 GHz network (which is by the way in accordance with the WIFI standard) and once it had a different one. So it is not a hardware limitation, but rather a software problem.

The use of the Discovery Server is not a really satisfactory solution, since in this case the exchange of data across multiple robots is - as far as I know - not possible.

Zakant, 2023-06-02:
When I tried to put the Turtlebot back into operation today, it had lost the connection to the WLAN. Reconnecting was not possible this time (SSID selectable, commission message is sent, but the platform does not connect and does not get assigned an IP). The setup is the same, which worked last time...
So it doesn't seem to be a sole problem with 2.4 or 5 GHz, but something else. I have attached the plattforms log files [here](https://github.com/turtlebot/turtlebot4/files/11632645/create3_logs2.txt).

hilary-luo, 2023-09-21:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-09-21 by hilary-luo

shamlian, 2023-09-21:
In the case you are still having a problem: the Create 3 team has released G.5.3 and H.2.3 to correct a bug in the connection workflow which may improve your experience. After updating your firmware, re-enter the network SSID, password, and region to refresh the robot's configuration, and please do let us know in https://github.com/iRobotEducation/create3_docs/issues/395 if it does not.