turtlebot/turtlebot4 #176: Issues with: Create 3 CPU load, TB4 freezing/restarting, ntpd - time way of, downgrading to Galactic?

tschinfra, 2023-05-24:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble

 **Expected behaviour**
- driving the TB4 around with the controller without freezing
- start mapping or navigating without problems

 **Actual behaviour**
1. From time to time the TB4 is freezing and you have to restart it (or it restarts itself). It often happens when pressing the "docking" button on the controller or when running SLAM with `ros2 launch turtlebot4_navigation slam.launch.py`.

**To Reproduce**
Provide the steps to reproduce:
1. updated the create 2 to VERSION: H.1.2
2. updated the Raspberry Pi image to Humble 1.0.0
3. connected a PS4 controller, because the original one does not work anymore since updating to Humble
4. configured the TB4 to discovery server

From now on i'm listing the current problems:
1. (this happened not every time) dock the TB4 with controller -> TB4 freezes -> Log message on create 3 `Server 'create-platform' crashed with exit code 134.`
2. (very often) CPU usage of the Create 3 close to 100% and because of this the robot freezes during SLAM or navigating
3. In the create 3 Logs all over the place is the message: `[INFO] (src/gateway_pinger.c:113) wifi is disconnected`
4. the time on the robot is way of, i tried restarting the ntpd with the create 3 beta features but it won't work. There is the following in the Log:
```
May 24 15:25:47 iRobot-9F44E8C9A8FE4225B49E96444FA8A979 user.notice ntpd: ntpd: bad address '0.irobot.pool.ntp.org'
May 24 15:25:47 iRobot-9F44E8C9A8FE4225B49E96444FA8A979 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 32s
```
I had the problems also before i switched to discovery server. There i had another log message telling something like 'time way of, ignoring ...' don't remember it clearly, because the Logs are cleared.

**Other notes**
I am on a relatively tight schedule due to a university project and need to get ahead. Would it be an option to switch back to Galactic? Have there been the same problems?

I am glad about any help.

Cur1ousKea, 2023-05-24:
Hi may i ask in which kind of project ure using it ? Im also currently on a Project for my university and struggeling as well with Humble, if u wanna hit me up on DC so we can help each other - FeelsDankMan#5663

ipa-rar, 2023-05-25:
@tschinfra @Cur1ousKea 
I have the same issue and I am waiting for the [update](https://github.com/turtlebot/turtlebot4/issues/147) from create3 base which will solve the issue with the CPU and RAM maxing out.

tschinfra, 2023-05-25:
@ipa-rar so i guess till then the best option is to downgrade to galactic to get going... I just can not wait much longer :/
Or would you suggest something else?

roni-kreinin, 2023-05-25:
@tschinfra The 100% CPU issues are related to Humble and are being tracked [here](https://github.com/iRobotEducation/create3_docs/issues/381). Switching back to Galactic is an option if you are on a tight schedule. You will have to update the Create 3 to the latest [firmware](https://iroboteducation.github.io/create3_docs/releases/g_4_5/), and flash the RPi4 SD card with the [0.1.3](http://download.ros.org/downloads/turtlebot4/) image. Instructions are available [here](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image).

sashag1991, 2023-07-24:
@tschinfra, did downgrade to galactic solved 100% CPU utilization ?

closed 2023-11-16 by hilary-luo