turtlebot/turtlebot4 #180: Turtlebot4 ros 2 navigation does not work no map

Markyyy55, 2023-05-25:
**Please provide the following information:**
 - OS: Ubuntu 22.04 Humble
 - ROS Distro: (e.g. Humble) 
 - Raspberry Pi new image Humble
 - Git repo from manual (did everything with terminal) 

 **Expected behaviour**
Normally when I start the ros2 navigation the turtlebot4 should scan the map and with the risview normally we should see the map and where the turtlebot4 is. 

 **Actual behaviour**
 I have a problem with the turtlebot4 the navigation does not work. So I did exactly what the manual told me to do. Normally you start to make the map and then can drive around to scan it. For visualization there is the command so that I can view where the turtlebot4 is. The problem is the view does not show anything. Just a grey box and a warning that no map received. 

So we wanted to make the discovery server so that there is a connection with create 3 and raspberry Pi. Also here we did everything what the manual told. But then after the option where we need to enter the ip and port of the discovery server, we just get this: rtnelink answers: no such process 

**To Reproduce**
Provide the steps to reproduce:
Watch the tutorial from YouTube about the ros2 navigation and see what happens

roni-kreinin, 2023-05-26:
Is your PC running in a VM?

Markyyy55, 2023-05-26:
Yes 
Windows 10 and then the Oracle VM

Markyyy55, 2023-05-29:
Is there any way to fix it?

roni-kreinin, 2023-05-29:
@Markyyy55 Can you check that you are able to ping the RPi from your VM? You may need to bridge your network if you cannot.

Markyyy55, 2023-05-31:
@roni-kreinin I can ping the raspberry pi. I used the Ip adress of the Turtlebot4 on the display and it worked. But I don’t understand why either the discovery server variant gives me: rtnetlink answers: no such process or why the ros 2 navigation option does not save a map and why I can’t see the map or the bot how he scans the map

roni-kreinin, 2023-06-01:
@Markyyy55 VM's can cause some issues with networking.

Can you share the output of the following commands from your VM:

- `ip a`
- `ip route`
- `ros2 topic list`

Markyyy55, 2023-06-02:
ok here are the outputs: 
ip a: 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp0s3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:3f:bc:0f brd ff:ff:ff:ff:ff:ff
    inet 10.0.2.15/24 brd 10.0.2.255 scope global dynamic noprefixroute enp0s3
       valid_lft 86246sec preferred_lft 86246sec
    inet6 fe80::323d:bf6:e3c:6fe7/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

ip route: default via 10.0.2.2 dev enp0s3 proto dhcp metric 100 
10.0.2.0/24 dev enp0s3 proto kernel scope link src 10.0.2.15 metric 100 
169.254.0.0/16 dev enp0s3 scope link metric 1000 

ros2 topic list: /parameter_events
/rosout

Markyyy55, 2023-06-02:
@roni-kreinin Yet I did not activity the network bridge

Markyyy55, 2023-06-21:
Hey @roni-kreinin We did it everything works now. Your suggestions worked all for us. Now we have another Question. We want to write some Scripts in Python. I would like to know how to do it and how can I start them, so that the turtlebot4 does what we write in the script? Could you give me some easy Examples?

roni-kreinin, 2023-06-21:
@Markyyy55 Try this tutorial https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_python.html.

Markyyy55, 2023-06-23:
ok worked fine. Now I have another Question. I found an example for a node from you with following name: followBot. Could you tell me how I can let it run on my turtlebot? In Addition i am Using Humble not galactic. Would there be some Changes?

roni-kreinin, 2023-06-23:
@Markyyy55 The `turtlebot4_examples` repo has not been released on Humble, but you can make a workspace and clone it. Build it with `colcon build --symlink-install`, then source your workspace and launch it:
```
ros2 launch turtlebot4_python_examples followbot.launch.py
```

Markyyy55, 2023-06-29:
@roni-kreinin Hey so did I understand it right? Can I run it on humble ? And for the case it won't work how can I run it on ros2 Humble? I already tried it like you described it above, but nothing happened or the package was not found. Could you help me out? How to write python code for ros2 humble? or are there any templates I could use?

roni-kreinin, 2023-06-30:
@Markyyy55 You will have to build the package locally:

```
mkdir ~/turtlebot4_ws/src -p
cd ~/turtlebot4_ws/src
git clone https://github.com/turtlebot/turtlebot4_examples.git
cd ~/turtlebot4_ws
colcon build --symlink-install
source ~/turtlebot4_ws/install/setup.bash
```

I would also recommend reading through the [ROS 2 tutorials](https://docs.ros.org/en/humble/Tutorials.html) to understand how to write python code for ROS 2.

Markyyy55, 2023-07-05:
@roni-kreinin Would be a big Help, because this is the only problem left. I tried to find a solution, but there were so many options, that I did not want to risk to break something.

roni-kreinin, 2023-07-07:
@Markyyy55 Are you building this on the robot PC? The depthai package should already exist on there. You can try running `rosdep install -r --from-paths src -i -y` from `~/turtlebot4_ws`.

closed 2023-09-21 by hilary-luo